FreeCAD/FreeCAD #16335: STL does not operate correctly for 3D printing

63Vette, 2024-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I created this simple design and used the STL import into AnkerMake Studio. The hole I had put did not show.

### Full version info

```shell
v.21.1
```

### Subproject(s) affected?

None

### Anything else?

As designed from FreeCAD
[image]

As imported STL from FreeCAD into AnkerMake
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-06:
I think this is a user error. You have probably selected all objects in the tree, You should only select the Body or its last feature for export.

See this Wiki page:
https://wiki.freecad.org/Std_Export

63Vette, 2024-09-06:
Can I send you the file and you provide instructions?

https://wiki.freecad.org/Std_Export

Roy-043, 2024-09-06:
Instructions have already been provided. What is unclear?

1. Open the file.
2. Select the Body (and only the Body) in the tree.
3. Invoke Std_Export.

63Vette, 2024-09-06:
Ok thank you for the assistance!

   
   - Open the file.
   - Invoke Std_Export.

luzpaz, 2024-09-06:
@63Vette you're using v0.21.x  
Most likely this issue you're reporting is user error. Please use forum.freecad.org to confirm your bug first before opening a ticket. 
Closing.

closed 2024-09-06 by luzpaz

63Vette, 2024-09-06:
I selected the body only from the tree and this is what exported, missing more of the structure!

This is the structure:

luzpaz, 2024-09-06:
@63Vette we can't see images when you respond via email. Better to respond directly on Github.

63Vette, 2024-09-06:
Ok