FreeCAD/FreeCAD #16339: Sketcher UI/UX, and possbily a bug too.

leoheck, 2024-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The line style can be improved a bit when Sketching, for instance. 

Both `construction mode` and the `non-construction mode ` (do we have a proper name for that?) start showing the same style until the last click is done.

First of all, I am using Dark Mode of OpenTheme.

Now, for instance, 

1. If I make a Line with the `construction mode`, it is drawn white. And then when I finish the line, it remains white.

2. Now, when I make a line with the `non-construction mode`, this same line is drawn white, and when it is done, the line turns into a dotted blue line. 

[UI/UX improvement] I understand that on item 2 it could/should also start being drown as dotted blue. However I don't know yet if this is not an OpenTheme issue.

[Possible bug] This may explain the reason that when designing a complex and huge project, the dotted blue lines take so much time to be drawn when editing an existing Sketch that has to update the list of the following features. 

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-06:
The color of the line as it is being drawn depends on a preference:
https://wiki.freecad.org/Sketcher_Preferences#Appearance (Creating line)

PaxTux, 2026-11-12:
@leoheck this feature was in the meantime implemented. Tested on 1.1.4 and weekly build. I think you can close this issue.
Also it seems that preference _Creating line_ mentioned by Roy-043 doesn't have any effect anymore.