FreeCAD/FreeCAD #16340: [UI/UX] PartDesign: Move Object after other Object

leoheck, 2024-09-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To find this menu: Right click on any object inside a PartDesign Body

First of all, this feature could be renamed to `Move Object Internally` since there are other operation named `Move Object to other Body`

When using the `Move Object after other Object`, a small popup with a drop-down list of Objects is displayed. 

[image]

The list of objects has just the name (Label) of the objects.
It could have the Object icon too, to make the process of finding this object easier. 
This will improve the usage specially when the order of this list does not reflect the order of the Objects Tree, which is almost always.

Took a picture of the lists since my screenshot tool to capture with a delay is not working right now.

[image]

My suggestion is to have something like this with colors, but using just a single level, of course.

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-07:
Maybe this should also be possible by drag and drop? @PaddleStroke

leoheck, 2024-09-07:
Ah, drag and drop would be interesting, indeed but it has a potential to mess things hard since things inside the Body have tight relationship.

PaddleStroke, 2024-09-09:
I think both Move "Object after other Object" and "Move Object to other Body" tools should be removed and these operations should be handled by drag and drop.
The drag and drop from and to a body just need to have the same rules as the current tools have. If the rules to accept are the same as current tools then I don't see why we would keep the tool no?

leoheck, 2024-09-09:
I agree, except that it is a bit hard to drag and drop things to the right place I want specially when this place is the last position of a nested object; the first position after a nested object; or even the last position of the object tree. This issue might may be due to the theme (OpenTheme) I am using, but I I feel it was like this since long time ago. This is also a hard thing to do when working on a huge list of objects and features.