FreeCAD/FreeCAD #16352: Sketcher: Can't zoom in

ChuckInGame, 2024-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Yes, this issue is only sometimes and it's hard to reproduce,
but i could isolate it to an example project, attached to this issue.

I open a sketch, this specific one, and I can't zoom into it.
At first, zooming works but will stop working before i reached the zoom level i need.

So the issue is:
* the sketch will not zoomed in well when i open the sketch
* when i try to zoom in, it works at the beginning, but then it stucks
* the view is not big enough to be able to read the constraint values

here a screencast:

https://github.com/user-attachments/assets/54ee5f17-a8fc-48ed-9738-0d72e46f01b6

[issue_05.zip](https://github.com/user-attachments/files/16917908/issue_05.zip)

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Render 2024.8.4
  * ThreadProfile 1.90.0
  * Help 1.0.3
  * Assembly4.backup1723411334.8566418 0.50.15 (Disabled)
  * freecad.gears 1.2.0
  * AirPlaneDesign 0.4.0
  * 3D_Printing_Tools
  * Assembly4 0.50.15
  * fasteners 0.5.26
  * MeshRemodel 1.9.23
  * pcb 6.2023.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-07:
This is caused by the perspective camera. There is a related Sketcher preference "Force orthographic camera when entering edit". I suggest you check that option. Note that "Restore camera position after editing" must also be checked.
https://wiki.freecad.org/Sketcher_Preferences#Display

ChuckInGame, 2024-09-07:
https://github.com/user-attachments/assets/97017249-e652-4e62-a8c8-98ddfce455cf

Here is a screencast where I can reproduce the issue inside the file "Topf_005."

When I try this in a new file, it works fine.

You can observe some unusual behavior:

I click on the plane, then go to Task -> Create Sketch. -> However, the task asks me again to select a plane, which normally shouldn't happen.

After clicking the plane again (as shown in the screencast), -> the sketch opens, but the visualization of the planes and origin is not as expected.

I draw a rectangle and try to zoom in, -> but it doesn't work properly.

When I add a constraint with a value, -> the constraint isn't shown while dragging. -> The resulting constraint has no visible value.

Roy-043, 2024-09-07:
Please read my previous comment.

ChuckInGame, 2024-09-07:
Hi
Roy-043,

sorry haven't seen your comment, because I wrote my last comment without refreshing it.

I changed it:
[image]

now, its better, the issue is gone, nice :)

but:
https://github.com/user-attachments/assets/e6408d40-11a7-4963-aea4-385dd7180802

As you can see, the sketch now opens zoomed in very deeply, and I have to scroll for several seconds to see the content.

Is there a solution for it ?

Roy-043, 2024-09-07:
This has to do with the previous zoom level:
1. Select the sketch before double-clicking it.
2. Press [Std_ViewFitSelection](https://wiki.freecad.org/Std_ViewFitSelection).
3. Double-click the sketch to enter edit mode.

ChuckInGame, 2024-09-07:
Yeah, many thanks :)
At least, not every guy knows all this, may be its better to improve the default behavior to avoid support request for things like that.

but ...works for me, thanks.

Regards,

Chuck

iplayfast, 2024-09-08:
I've found that if my computer suspends when I go back to it, everything still works except the zoom using the mouse scroll wheel. Exiting and restarting FreeCAD is the only way of fixing that I know of. Maybe related or not.

risacher, 2024-11-25:
Thanks @Roy-043 .   I had this problem also (both 1.1.0dev and 1.0.0), and your suggested fix/workaround resolved it.  I agree with @ChuckInGame suggestion that maybe this should be default behavior.

maxwxyz, 2024-12-08:
@Roy-043 what to do with that issue?

Roy-043, 2024-12-09:
The 2nd issue (zoom level when opening a sketch) could become a new  feature request. The initial issue has been addressed I think.

closed 2024-12-09 by maxwxyz