FreeCAD/FreeCAD #16353: Draft: 1.0RC Draft_Hatch displays pattern in a wrong way

semhustej, 2024-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using .pat files to create hatch pattern in **FreeCAD 1.0RC1**, I encountered FreeCAD displaying some patterns in a wrong way. Two patterns are provided as samples.
Simple patterns seem to be displayed OK, more complex patterns seem to be displayed in a wrong way. Two examples follow.

How patterns are displayed in FreeCAD (incorrect)
[image]

How patterns are displayed in BricsCAD (DWG Editor) and Revit (correct way)
[image]

Provided as a zipped attachement:

1. FreeCAD file
2. DXF file with correct displaying of patterns
3. Two pattern files used

[FC-pattern-issue-files.zip](https://github.com/user-attachments/files/16917834/FC-pattern-issue-files.zip)

The only discussion about the topic I found is this one, but it seems to be out of date. I am not sure, whether more complex patterns (such as those provided) will not be supported, as the .pat format seems to be not too popular.
https://forum.freecad.org/viewtopic.php?t=68205

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Draft

### Anything else?

The behavior is not a regression. When trying to use these .pat files in **FreeCAD 0.21.2,** the command crashes with this output:

_13:08:42  <Exception> DGH::getTrimmedLines - baseFactory failed
13:08:43  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\draftobjects\hatch.py", line 124, in execute
    shape = TechDraw.makeGeomHatch(face, obj.Scale, obj.Pattern, obj.File)
<class 'RuntimeError'>: FreeCAD exception thrown (DGH::getTrimmedLines - baseFactory failed)
13:08:43  Recompute failed!
13:08:43  <Exception> DGH::getTrimmedLines - baseFactory failed
13:08:43  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\draftobjects\hatch.py", line 124, in execute
    shape = TechDraw.makeGeomHatch(face, obj.Scale, obj.Pattern, obj.File)
<class 'RuntimeError'>: FreeCAD exception thrown (DGH::getTrimmedLines - baseFactory failed)
13:08:43  Recompute failed!
13:09:15  <Exception> DGH::getTrimmedLines - baseFactory failed
13:09:15  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\draftobjects\hatch.py", line 124, in execute
    shape = TechDraw.makeGeomHatch(face, obj.Scale, obj.Pattern, obj.File)
<class 'RuntimeError'>: FreeCAD exception thrown (DGH::getTrimmedLines - baseFactory failed)
13:09:15  Recompute failed!
13:09:15  <Exception> DGH::getTrimmedLines - baseFactory failed
13:09:15  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\draftobjects\hatch.py", line 124, in execute
    shape = TechDraw.makeGeomHatch(face, obj.Scale, obj.Pattern, obj.File)
<class 'RuntimeError'>: FreeCAD exception thrown (DGH::getTrimmedLines - baseFactory failed)
13:09:15  Recompute failed!
13:09:22  <Exception> DGH::getTrimmedLines - baseFactory failed
13:09:22  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\draftobjects\hatch.py", line 124, in execute
    shape = TechDraw.makeGeomHatch(face, obj.Scale, obj.Pattern, obj.File)
<class 'RuntimeError'>: FreeCAD exception thrown (DGH::getTrimmedLines - baseFactory failed)
13:09:22  Recompute failed!
13:10:00  ----------------
13:10:00  GuiCommand: Rectangle
13:10:00  Pick first point
13:10:00  Pick opposite point
13:10:21  <Exception> DGH::getTrimmedLines - baseFactory failed
13:10:21  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\draftobjects\hatch.py", line 124, in execute
    shape = TechDraw.makeGeomHatch(face, obj.Scale, obj.Pattern, obj.File)
<class 'RuntimeError'>: FreeCAD exception thrown (DGH::getTrimmedLines - baseFactory failed)
13:10:21  Recompute failed!
13:10:21  <Exception> DGH::getTrimmedLines - baseFactory failed
13:10:21  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\draftobjects\hatch.py", line 124, in execute
    shape = TechDraw.makeGeomHatch(face, obj.Scale, obj.Pattern, obj.File)
<class 'RuntimeError'>: FreeCAD exception thrown (DGH::getTrimmedLines - baseFactory failed)
13:10:21  Recompute failed!_

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-07:
Dashed lines in patterns are not yet supported. But that explains only some of what we see. I'll  have to check the rest.

This is a Draft issue, but also a TechDraw issue as we rely on TD for the generation of the pattern.

semhustej, 2024-09-07:
Thanks for the clarification. Not supporting dashed lines in patterns is useful information. This explains one part of the problem. 
The other problem seems to be offset of some of the lines.

semhustej, 2024-09-13:
Not sure whether fix was issued but I tested with new release of RC1.38684 and the offset of the patterns seems to be fixed. See screenshot:
[image]

The lines are not dashed, but they are placed in the right positions.

FC version:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/

semhustej, 2024-10-28:
New finding (I am not talking about the dashed lines):

The pattern of the hatch changes based on applied scale. Apparently TechDraw_GeometricHatch has the same problem, I am able to reproduce it with TechDraw_GeometricHatch too. It is described in a discussion here: https://forum.freecad.org/viewtopic.php?t=91711

TechDraw_GeometricHatch scale 0.028
[image]

TechDraw_GeometricHatch scale 0.027
[image]

It seems Draft_Hatch and TechDraw_GeometricHatch might have the same issue.

luzpaz, 2024-10-30:
Cc @WandererFan

closed 2025-03-03 by chennes

semhustej, 2025-04-23:
Tested both `Draft_Hatch` and `TechDraw_GeometricHatch`. Both work.

Thanks a lot, this is a great improvement!