FreeCAD/FreeCAD #16359: TechDraw: Balloon annotations move both annotation and tip even with simple drag

johnsonm, 2024-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Even without control-click, the whole arrow moves when dragging the annotation.

Previously, a simple click-drag would drag the annotation, leaving the tip in the same place; a control-click-drag would draw the whole annotation including moving the arrow tip.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38644 (Git)
Build type: Unknown
Branch: main
Hash: 7ed1e9380acccd686a0e3573f883ac793aec3299
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.5.24
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

johnsonm, 2024-09-07:
double-click drag on the annotation is now the workaround I've found

WandererFan, 2024-09-08:
I thought we already had an issue for this, but maybe it was a forum topic.

What's happening is that the multi-select option is adding a ctrl modifier to mouse events, since that is the only way QGraphicsScene will allow multiple object selection.

If you turn off multiple selection in the General tab of preferences, you'll get the old behaviour.

There is a mod in development that will allow the user to pick a key combination for balloon origin drag.

johnsonm, 2024-09-10:
This appears fixed in current `main` — at least, I don't reproduce it after updating to this version:

OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38693 (Git)
Build type: Unknown
Branch: main
Hash: 85ade21436d4ce618691720688b299c9b6902590
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.5.24
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0

johnsonm, 2024-09-10:
Sorry, I hadn't seen your comment when I closed the issue. In any case, I didn't change the multiple selection setting — I had disabled it as a result of that prior conversation, and it wasn't enabled when I reported this problem. In any case, it seems to have gone away now.

Maybe it was a bug somewhere else in selection that was fixed in the meantime.

closed 2024-09-10 by johnsonm