FreeCAD/FreeCAD #16360: [UI/UX] Orbit selector on Status bar not set correctly after fresh install

inkapnea, 2024-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On a fresh installation the Orbit selector on the Status Bar is set to **Turntable**,
while in the Preferences Editor it's set to **Trackball**. The later corresponds to the the actual behavior.

Status bar: Turntable
Preferences: Trackball

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

&

[code]
OS: KDE Flatpak runtime (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

&

Weekly Build on Windows 11
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-07:
Can you show what you mean via screenshot ? TIA

inkapnea, 2024-09-08:
> Can you show what you mean via screenshot ? TIA

[image]
[image]

To replicate the issue, prior to launching FreeCAD, I deleted all the files in the folders:

/home/username/.config/FreeCAD/
-on Linux, for Weekly Dev 0.22.0dev Appimage.

/home/username/.var/app/org.freecadweb.FreeCAD/config/FreeCAD/
-on Linux, for Stable 0.21.2 flathub version.

C:\Users\Foxberry\AppData\Roaming\FreeCAD\
-on Windows, for 0.21.2 Stable installed version.

And it happens every time, on both systems and on both Stable and Dev FreeCAD versions.

luzpaz, 2024-09-08:
@Rexbas any chance you can take a look at this ?

Rexbas, 2024-09-08:
Yes

PR #16716 (2024-09-22): Tux: Fix default orbit style shown in status bar
Fixes #16360.
closed 2024-09-22 by chennes