FreeCAD/FreeCAD #16361: Coincedent to Extranal Geometry is not sticking. RC build error

iplayfast, 2024-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After pulling in an external geometry, and using a coincedent of the end of a line to a point, the end of the line jumps to the correct spot, however the line can still be moved after that. It should be stuck to the point

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Unknown
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/Canada (en_CA)
Installed mods: 
  * freecad.gears 1.0.0
  * 3D_Printing_Tools
  * CycloidGearBox
  * FeedsAndSpeeds 0.5.0
  * Ondsel-Lens 2024.1.22.01
  * ExplodedAssembly
  * Nodes 0.1.36
  * 3DfindIT 1.2.0
  * A2plus 0.4.63
  * MeshRemodel 1.9.22
  * lattice2 1.0.0
  * fasteners 0.5.10
  * Assembly3 0.12.2
  * Assembly4 0.50.7
  * POV-Ray-Rendering
  * PieMenu 1.2.7
  * CurvedShapes 1.0.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

iplayfast, 2024-09-08:
So what I think happened was I moved or deleted a pad that I was taking an external from, Then later went and edit the sketch and the externals were still there but could not be attached to.
This is probably not a bug, but more of a FreeCad thing. If a pad is deleted I would expect any references to it to also be deleted. but FreeCAD doesn't clean up external references when something is deleted.

luzpaz, 2024-09-08:
@iplayfast would you mind making a screencast of the bug ?

lagnat, 2024-09-09:
I might be seeing the same issue.  Set several constraints, goes fully resolved, exit sketcher.  Go directly back into the sketch and the constraints are gone.  Not just coincident constraints.

lagnat, 2024-09-09:
On second glance, I see what is happening.  The external geometry is broken but the sketcher doesn't indicate that.  In past versions, the external geometry name would show with question marks and the geometry itself would show with a different color however with the latest weekly builds that is no longer the case.  The sketcher is happy to allow you to set constraints on a broken external geometry until you exit the sketcher at which point the report view indicates that the external geometry is broken and the constraints are getting removed.

maxwxyz, 2024-09-09:
Please provide an example file and steps how to reproduce. Ideally record a macro of what you are doing and share it here.

lagnat, 2024-09-09:
[Constraint.FCStd.zip](https://github.com/user-attachments/files/16935668/Constraint.FCStd.zip)

https://github.com/user-attachments/assets/cfa6f140-f329-43aa-aef1-a1df63191f81

DerMouse, 2024-09-15:
> I might be seeing the same issue. Set several constraints, goes fully resolved, exit sketcher. Go directly back into the sketch and the constraints are gone. Not just coincident constraints.

Seems I have the very same problem - except I cannot find any breakage on the Corps body, used as an external geometry for Lid.

Working on Lid/Sketch004, I can add constraints (here, some distance constraint)
[image]

But as soon as I leave the sketch, I get a "Missing external geometry reference" error and all constraints are removed
[image]

Here is the zipped FCStd file, in case it could be useful.
[MissingGeometry.zip](https://github.com/user-attachments/files/17005578/MissingGeometry.zip)

DerMouse, 2024-09-15:
Forgot some data :

OS: Gentoo Linux
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.2.0
  * persistenttoolbars

maxwxyz, 2024-09-15:
@PaddleStroke @bgbsww any idea here?

bgbsww, 2024-09-15:
@AjinkyaDahale is a useful set of eyes on sketcher geo concerns.  I'll take a look myself later.

bgbsww, 2024-09-15:
If you look at the Constraints property of the sketch, it is empty after you close the sketch.  So the constraints were never recorded, as opposed to not being read in correctly when you open.

bgbsww, 2024-09-15:
Okay, so this doesn't immediate replicate with simple sketches and external geometry.  Looking at the example file and actions above, we get:

```
19:43:37  <Sketch> SketchObject.cpp(9136): External geometry MissingGeometry#Sketch004.e72 missing reference: Binder.;Edge71;:Hb90,E;:Hb69:8,E
19:43:37  <Sketch> SketchObject.cpp(9136): External geometry MissingGeometry#Sketch004.e73 missing reference: Binder.;Edge91;:Hb90,E;:Hb69:8,E
19:43:37  <Sketch> SketchObject.cpp(9136): External geometry MissingGeometry#Sketch004.e89 missing reference: Binder.;#387:5;:Hb7e,E;:Hb8a,E;:Hb69:8,E
19:43:37  <Sketch> SketchObject.cpp(9136): External geometry MissingGeometry#Sketch004.e90 missing reference: Binder.;#385:5;:Hb7e,E;:Hb8a,E;:Hb69:8,E
19:43:37  <Sketch> SketchObject.cpp(9136): External geometry MissingGeometry#Sketch004.e91 missing reference: Binder.;#37e:5;:Hb7e,E;:Hb8a,E;:Hb69:8,E
19:43:37  Missing external geometry reference
Clear constraints to external geometry
```
And what you can see is that something involving specifically references to binders is failing, and then sketch is recovering by removing that geometry.

So this appears to be restricted to cases with binders unless someone has other evidence.

lagnat, 2024-09-16:
The file I uploaded doesn't use binders.  The issue is created by creating an external geometry reference on a chamfer edge and then deleting the chamfer.

bgbsww, 2024-09-16:
Fair enough ... the file from DerMouse does though.    There are no reproduction steps given for Constraint.FCStd, so it is not usable for me.  [mov files do not render correctly on my platform, which is part of why we ask for steps and ideally a macro]

lagnat, 2024-09-16:
The video doesn't show how to setup the issue, only the behavior after the external geometry is in a broken state.  Try this and let me know if the Macro would still be helpful.  https://www.youtube.com/watch?v=2xj-yLeue6g

bgbsww, 2024-09-16:
Thanks, that will allow me to replicate!  To answer your question though,
yes, a macro is *always* preferable, at least for me, because I can convert
it easily into a test, and then start trying to trace the code.  If I have
to watch and pause a video, recreate the steps to just get to that stage,
and then proceed it takes a lot more effort.

bgbsww, 2024-09-16:
So, this appears actually to have little to do with constraints.  I would call it "Sketcher: invalid External Geometry is shown in Sketcher".  Everything else follows from that.  There is a SketchObject::validateExternalLinks method, but it looks only in the ExternalGeometry property and not the ExternalGeo where the links to deleted geometry reside.  That then allows you to see and use that geometry in a sketch, but when you exit, it is detected as invalid in constraints and removed.

We are missing something that synchronizes the ExternalGeo, which might be missing code that didn't move over.

Easy replication steps:
1) Make a sketch
2) Draw something in it, like a box.
3) Exit sketcher
4) Make a second sketch
5) Add one line of external geometry from the first sketch
6) Add a line
7) Add a coincident constraint between one of the points in the external geometry, and one of the points on your line
8) Exit sketcher
9) Open first sketch
10) Delete the geometry referred to by the second sketch
12) Close sketch.  Note deleted geometry is not shown
13) Open second sketch.   Note deleted geometry has come back from the dead and is now a zombie about to eat your neck.
14) Notice constraint is gone.  Notice you can add it, but everytime you close the sketch it will be removed.

bgbsww, 2024-09-16:
After some discussion on discord, @kadet1090 made the excellent recommendation that we should make two changes here:

1) Allow constraints on external geometry references to deleted objects to "stick" since we have the geometry, even if it is no longer attached to that external geometry

2) Make it visually clear that a piece of external geometry has become detached (  ExternalGeometryExtension::Missing is set ) by drawing it in red, adding a broken link icon in the geometry list, and tweaking the tooltips and labels in a consistent way so the user knows what is happening

3) Potentially offer some way to set replacement external geometry.  That could be a separate effort.

lagnat, 2024-09-16:
> 3. Potentially offer some way to set replacement external geometry.  That could be a separate effort.

Regarding this point.  @realthunder has this feature in his fork and it's absolutely fantastic.  I can't stress enough how much pain this feature removes.

bgbsww, 2024-09-16:
@FreeCAD/design-working-group I clearly stumbled into an issue that you are thinking hard about and working through concepts.  Title change again:  "Sketcher: Design approach to missing External Geometry References"

I'm going to leave this lying now, and when we have consensus on the right way to address this, someone can pick the bug up.  The early common thought seems to be that broken references for which we have geometry should be left in place in the sketcher, and allowed to be used to make constraints that stay around.  However, how to display that geometry and the tree and what options to allow the user are currently under discussion.

DerMouse, 2024-09-16:
> So, this appears actually to have little to do with constraints. I would call it "Sketcher: invalid External Geometry is shown in Sketcher". Everything else follows from that. There is a [...]

I'm totally out of my depth here, so my question could rate a 10 on the 0-to-5 Noob's scale, but... 

In the MissingGeometry file attached to https://github.com/FreeCAD/FreeCAD/issues/16361#issuecomment-2351552331, is the first body (used as binder for the second body) invalid ? I did all checks I could think of, and didn't find anything wrong. If someone could tell me what is wrong (or at least point me into the proper direction)...

Thanks.

bgbsww, 2024-09-16:
It just looks to me like the external geo in Sketch004 isn't right.  I would delete and re-add all of them and see if the issue goes away.  That would be caused because you had added them, and then did something to the object they refer to, and that broke those links.  Let us know!

DerMouse, 2024-09-17:
> It just looks to me like the external geo in Sketch004 isn't right. I would delete and re-add all of them and see if the issue goes away. That would be caused because you had added them, and then did something to the object they refer to, and that broke those links. Let us know!

You are right (I did not doubt that). Long story short, the first body was indeed different somewhere back in time, and then I dropped some sketches from it (after creating the second body and binder). So your hypothesis about where this mess came from is validated.

In the Lid body, sketch by sketch starting with Sketch004, I removed all external constraints and re-added them, closed the sketch, reopened it to check if the constraints stuck (yes !), and went to the next sketch. At one point, Fc was very unhappy (see attached screen shot) but nothing really unexpected by then.

[image]

Cleaned up all sketches, all constraints are sticking now. I've learned something new, the day will not be wasted.

Thanks for the hint on the model, and a lot of thanks for all the incredible work of everyone on FC.

PR #16590 (2024-09-18): Sketcher:  Do not consider missing external geometry to be an error; highlight i…
…n conflict color.  Fix #16361
closed 2024-09-18 by chennes