FreeCAD/FreeCAD #16362: Opening Addons manager freezes freecad on AppImage.

Garbg, 2024-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you open the addons manager it freezes freecad. I opened the app image in the terminal and the following error was printed in the terminal.
```
Traceback (most recent call last):
  File "/tmp/.mount_FreeCA93eBJ0/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 455, in run
    repo.load_metadata_file(repo_metadata_cache_path)
  File "/tmp/.mount_FreeCA93eBJ0/usr/Mod/AddonManager/Addon.py", line 375, in load_metadata_file
    metadata = MetadataReader.from_file(file)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCA93eBJ0/usr/Mod/AddonManager/addonmanager_metadata.py", line 255, in from_file
    return MetadataReader.from_bytes(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCA93eBJ0/usr/Mod/AddonManager/addonmanager_metadata.py", line 260, in from_bytes
    element_tree = ET.fromstring(data)
                   ^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCA93eBJ0/usr/lib/python3.11/xml/etree/ElementTree.py", line 1350, in XML
    parser.feed(text)
xml.etree.ElementTree.ParseError: mismatched tag: line 31, column 6
```

### Full version info

```shell
Latest weekly build with filename as of September 7th 2024
FreeCAD_weekly-builds-38643-conda-Linux-x86_64-py311.AppImage

OS: openSUSE Tumbleweed (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Addon Manager

### Anything else?

I'm not sure if this only affects appimages or not.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-08:
Cannot confirm, using the same AppImage, AddonManager loads successfully, here is the Report View output:

```
12:02:34  Checking network connection...
12:02:34  GitHub's zen message response: Mind your words, they are important.
12:02:34  Attempting to normalize non-compliant license 'GPL-2'... found valid SPDX license ID GPL-2.0
12:02:34  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
12:02:34  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
12:02:34  Attempting to normalize non-compliant license 'AGPLv3.0'... found valid SPDX license ID AGPL-3.0
12:02:34  Attempting to normalize non-compliant license 'GPLv3'... found valid SPDX license ID GPL-3.0
12:02:35  Attempting to normalize non-compliant license 'LGPLv3'... found valid SPDX license ID LGPL-3.0
12:02:35  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
12:02:35  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
12:02:35  Attempting to normalize non-compliant license 'lgpl-2.1-or-later'... found valid SPDX license ID LGPL-2.1-or-later
12:02:35  Attempting to normalize non-compliant license 'LGPLv3'... found valid SPDX license ID LGPL-3.0
12:02:35  Attempting to normalize non-compliant license 'AGPLv3.0'... found valid SPDX license ID AGPL-3.0
12:02:35  Attempting to normalize non-compliant license 'LGPL-3'... found valid SPDX license ID LGPL-3.0
12:02:35  Attempting to normalize non-compliant license 'GPLv3.0'... found valid SPDX license ID GPL-3.0
12:02:35  Attempting to normalize non-compliant license 'GPLv2.1'... failed to normalize (typo in ID or invalid version number??)
12:02:35  Attempting to normalize non-compliant license 'GPL-3'... found valid SPDX license ID GPL-3.0
12:02:35  Attempting to normalize non-compliant license 'AGPLv3.0'... found valid SPDX license ID AGPL-3.0
12:02:35  Attempting to normalize non-compliant license 'CCOv1'... failed to normalize (typo in ID or invalid version number??)
12:02:35  Attempting to normalize non-compliant license 'GPL-3'... found valid SPDX license ID GPL-3.0
12:02:35  Attempting to normalize non-compliant license 'GPL-3'... found valid SPDX license ID GPL-3.0
12:02:35  Invalid version specified for tag version in Addon Trails: 2022-01
12:02:35  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
12:02:35  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
12:02:38  Attempting to normalize non-compliant license 'GPL-3'... found valid SPDX license ID GPL-3.0
```

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.66
```

Garbg, 2024-09-08:
I don't know what's up with it then. Maybe mine is somehow corrupted or something? I'll try redownloading it.

Syres916, 2024-09-08:
I would hazard a guess that the AddonManager cache folder needs a clear out, on Linux it's usually held in:
/home/_username_/.cache/FreeCAD/Cache

Garbg, 2024-09-08:
> I would hazard a guess that the AddonManager cache folder needs a clear out, on Linux it's usually held in: /home/_username_/.cache/FreeCAD/Cache

Yup, that fixed it.

closed 2024-09-08 by Garbg

luzpaz, 2024-09-08:
@Syres916 nice call.  Is this documented somewhere perhaps ?

Syres916, 2024-09-10:
> @Syres916 nice call. Is this documented somewhere perhaps ?

I can't find it documented, for Post 1.0 I was going to create a `Clear Cache` button in _Preferences_ -> _Addon Manager_ next to the _Cache Update_ combobox which even accounting for translations seems rather wide so it shouldn't look cramped. This can then be added to the corresponding Wiki page and IMHO be a more polished solution. @chennes would this be acceptable?

chennes, 2024-09-10:
Sure -- you could also consider putting it in preferences, since it's not expected that it will get used very often. Or as a drop-down option off of the "Check for Updates" button. The cache process is going to change quite a lot in 1.1, so that would be a useful debugging tool anyway.