FreeCAD/FreeCAD #16363: BIM: strange behavior when computing hatches

gianlucafiore, 2024-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a group of closed walls, the entire interior is completely filled, which should not be the case.
And in addition, the hatches on the walls lean in the opposite direction. I am using the Autocad hatch template.

[image]

[edificio nuevo 2.zip](https://github.com/user-attachments/files/16920125/edificio.nuevo.2.zip) (test file)
[acad.zip](https://github.com/user-attachments/files/16920127/acad.zip) (.pat)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-08:
The Hatch command just hatches the faces from the source object. So this is caused by the Shape2DView object. There are 2 overlapping rectangular faces, instead of one face with a hole, in the problem area.

The attached file shows a potential workaround:
1. Create a Shape2DView without faces.
2. Create a PartDesign_SubShapeBinder from that object.
3. Hatch the created Binder.

[Shape2DView-faces-issue.FCStd.zip](https://github.com/user-attachments/files/16921717/Shape2DView-faces-issue.FCStd.zip) (remove the zip extension)

yorikvanhavre, 2024-09-09:
Generally speaking there are indeed still a few of those things not handled in the Draft Shape2DView tool. The problem is here, with the current system ( https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Draft/draftobjects/shape2dview.py#L317 ), when a BIM model is cut by a section plane, a series of cut faces are produced and compounded together. I guess instead of just adding the produced faces to the stack, we should first see if one is inside the other and do the necessary cuts. Problem is, how to efficiently detect that one face is inside another.

paullee0, 2025-01-20:
I have a quick test, seem Shape.generfuse(SectionPlane) return result with Face of correct result.  

Not sure if Shape.section, Shape.slice(s) are simpler approaches.

Any comments?

[image]
[image]

[Test_ Shape2dView_ 01 .FCStd.zip](https://github.com/user-attachments/files/18480891/Test_.Shape2dView_.01.FCStd.zip)

paullee0, 2025-01-20:
Further tests.

Shape.fuse(SectionPlane) not works
Shape.common(SectionPlane) return Shape of correct Face - seem simpler than Shape.generalFuse()

[image]
[image]

[Test_ Shape2dView_ 02.FCStd.zip](https://github.com/user-attachments/files/18481073/Test_.Shape2dView_.02.FCStd.zip)

paullee0, 2025-01-25:
> Generally speaking there are indeed still a few of those things not handled in the Draft Shape2DView tool. The problem is here, with the current system ( [...]

The problem occurs when the selected ProjectionMode is Cutfaces, right?

https://github.com/FreeCAD/FreeCAD/blob/29ea277b3ba4f38ce7dd0a51ca0d4e6e03b65cf2/src/Mod/Draft/draftobjects/shape2dview.py#L318

paullee0, 2025-01-25:
OK, draft revision below for review and comment.

Currently CutFaces works with 'In Place';  without 'In Place', more study is required.

[shape2dview.py.txt](https://github.com/user-attachments/files/18548226/shape2dview.py.txt)

[image]
[image]

[Test_ Shape2dView_ 02_ r1.FCStd.zip](https://github.com/user-attachments/files/18548230/Test_.Shape2dView_.02_.r1.FCStd.zip)

paullee0, 2025-01-26:
Tested with the OP model file, with Translate set to False, used FC internal hatch pattern.

[image]

[edificio nuevo 2_ r2.FCStd.zip](https://github.com/user-attachments/files/18548555/edificio.nuevo.2_.r2.FCStd.zip)

paullee0, 2025-01-26:
Revised draft attached - [shape2dview.py.txt](https://github.com/user-attachments/files/18548746/shape2dview.py.txt)

Now, when CutFaces + InPlace=False, the problem is self.getProjected() / TechDraw.projectEx(shape, direction) make all the correct Faces projected into lines/wires only.   (With CutFaces + InPlace=True, we now have correct faces)

Anyone has idea how this getProjected() could be made by simple translation / move / rotate (for cutFaces), thanks :) 

[image]

paullee0, 2025-01-26:
Further refined - tackled when any solid is not cut:-  [shape2dview.py.txt](https://github.com/user-attachments/files/18550772/shape2dview.py.txt)

Still looking way to project faces as TechDraw.projectEx() does.

[Test_ Shape2dView_ 02_ r3 .FCStd.zip](https://github.com/user-attachments/files/18550782/Test_.Shape2dView_.02_.r3.FCStd.zip)

[image]

semhustej, 2025-01-27:
I did a quick test with the library from the last post form @paullee0 and it works fine:

[image]

paullee0, 2025-01-27:
> I did a quick test with the library from the last post form [@paullee0](https://github.com/paullee0) and it works fine:
> 

Thanks for the testing.  The problem left now is find how to 'project' the faces as TechDraw.projectEx() does.

paullee0, 2025-01-28:
@yorikvanhavre  I am able to produce face with 'Closed loop ArchWall' as shown above, but now the problem is the Face is created 'In Place'.

When user select 'In Place' to be False, I have no idea what the original "TechDraw.projectEx()"  -  _groups = TechDraw.projectEx(shape, direction), do to 'project' the edges, but as WanderFan point out in FC Forum, it only deal with edge not face.  WanderFan has not much idea about this function.  It appears to me it is more or less a translation?

Any idea about the algorithm behind and maybe alternative code could be developed to 'project' the face.

Thanks.

paullee0, 2025-01-28:
OK, I found a slightly convoluted way for 'In Place = False' case to create face, see if the behaviour is consistent with the old way which did not preserve face :)

Thanks.

[shape2dview.py.txt](https://github.com/user-attachments/files/18576940/shape2dview.py.txt)

[Test_ Shape2dView_ 02_ r3.FCStd.zip](https://github.com/user-attachments/files/18576960/Test_.Shape2dView_.02_.r3.FCStd.zip)

[image]

paullee0, 2025-01-28:
OK, this revision should works with Face from closed ArchWall :)

[shape2dview.py.txt](https://github.com/user-attachments/files/18577207/shape2dview.py.txt)

(same test model file)

[image]

paullee0, 2025-01-29:
PR submitted to fix this -

https://github.com/FreeCAD/FreeCAD/pull/19306

closed 2025-02-05 by yorikvanhavre

paullee0, 2026-04-18:
Further cornercase and draft PR :)

- https://forum.freecad.org/viewtopic.php?p=884294#p884294