FreeCAD/FreeCAD #16364: Draft Array: When expand array is enabled, changing center of rotation has no effect

theosib, 2024-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I made an Draft Array of Std_Part so that I could see how four of the same part would fit together in a design. This way, I can print four of the same thing and get them to fix together. In order to check the alignment and hide/show individual faces, I enabled "expand array," but I also need to adjust the center of rotation to tweak the alignment. But when expand array is set, changing the center of rotation on the array object has no effect. I have to set expand array to false, tweak the center, and then set it to true again to get the expected result.
[lampshde.FCStd.zip](https://github.com/user-attachments/files/16920137/lampshde.FCStd.zip)

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-08:
This is probably intended behavior. There is a property "Always Sync Placement" that should be set to true to achieve what you want. But doing so causes "still touched after recompute" errors.

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

Roy-043, 2024-09-08:
Closing this in favor of #16368.

closed 2024-09-08 by Roy-043