FreeCAD/FreeCAD #16365: Part Cut failing although geometry of constituents shows no errors

theosib, 2024-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In this design, please see Cut002, which is "left pipe" minus "mid cut001". The all the geometries check out. So the result should be "left pipe" with a chunk cut out of it. But instead, I still get a complete piece.

So the first image is the pipe. 
[image]

The second image is a chunk I want to cut from it. 
[image]

Instead of getting a piece cut out, there are just new contours added. 
[image]

[lampshde.FCStd.zip](https://github.com/user-attachments/files/16920167/lampshde.FCStd.zip)

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-09-08:
I tried setting "refine" on each of the parts. Didn't help.

theosib, 2024-09-08:
I tried simplifying things a bit. 

[image]

"left pipe" is a simple extruded sketch minus another extruded sketch. "top cut001" is also just an extruded sketch. The geometry on both checks out fine.

But when I try to do the intersection of the two ("top of left pipe"), I get an invalid compound. 
[image]

I'm not doing anything super complicated here, so I can't figure out why it's failing.

FEA-eng, 2025-10-17:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * A2plus 0.4.68
  * freecad.gears 1.3.0
```

FEA-eng, 2026-11-05:
I don't have recomputation errors anymore, just this:

```
18:29:42  <PropertyLinks> PropertyLinks.cpp(445): recovered drifted element reference Edge1 -> Edge1 in lampshde#Extrude003
18:29:42  <PropertyLinks> PropertyLinks.cpp(445): recovered drifted element reference Edge4 -> Edge4 in lampshde#Extrude003
18:29:42  <PropertyLinks> PropertyLinks.cpp(445): recovered drifted element reference Edge2 -> Edge2 in lampshde#Extrude003
18:29:42  <PropertyLinks> PropertyLinks.cpp(445): recovered drifted element reference Edge3 -> Edge3 in lampshde#Extrude003
18:29:42  <PropertyLinks> PropertyLinks.cpp(445): recovered drifted element reference Face1 -> Face1 in lampshde#Extrude003
```

Then it looks like that:

[image]

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```