FreeCAD/FreeCAD #16367: WinError 193 %1 is not a valid Win32 application

corpsedog, 2024-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a new installation of FreeCad downloaded within the last two weeks (I am a new user following the examples in the FreeCad manual). Opened FreeCad, selected Create new, then launched Part Design workbench. Following messages then appeared:
21:53:09  [WinError 193] %1 is not a valid Win32 application
21:53:09  Traceback (most recent call last):
  File "<string>", line 41, in Initialize
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Program Files\FreeCAD 0.21\Mod\PartDesign\WizardShaft\WizardShaft.py", line 28, in <module>
    from .Shaft import Shaft
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Program Files\FreeCAD 0.21\Mod\PartDesign\WizardShaft\Shaft.py", line 25, in <module>
    from .SegmentFunction import SegmentFunction,  IntervalFunction,  StressFunction,  TranslationFunction
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Program Files\FreeCAD 0.21\Mod\PartDesign\WizardShaft\SegmentFunction.py", line 25, in <module>
    import numpy as np
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\John\AppData\Roaming\Python\Python38\site-packages\numpy\__init__.py", line 140, in <module>
    from . import _distributor_init
  File "C:\Program Files\FreeCAD 0.21\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\John\AppData\Roaming\Python\Python38\site-packages\numpy\_distributor_init.py", line 26, in <module>
    WinDLL(os.path.abspath(filename))
  File "C:\Program Files\FreeCAD 0.21\bin\lib\ctypes\__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-08:
> Word size of FreeCAD: 64-bit

Does 0.21 run on win32 ?

Syres916, 2024-09-08:
@corpsedog you don't say whether you previously had an old FreeCAD version in the past. The reason I bring that up is there was a previous report and solution on the forum, see https://forum.freecad.org/viewtopic.php?p=427200#p427200