FreeCAD/FreeCAD #16369: Unable to put breakpoint in executable binary

mdrost, 2024-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,
I compiled FreeCAD on Windows with enabled `FREECAD_RELEASE_PDB` flag and noticed that I am unable to put breakpoints in executable part of the code (for example in `Main/MainGui.cpp`) while in shared libraries it works fine.
When in `SetGlobalCompilerAndLinkerSettings.cmake` I add line `set (CMAKE_EXE_LINKER_FLAGS_RELEASE "${CMAKE_EXE_LINKER_FLAGS_RELEASE} /DEBUG")` under `if(FREECAD_RELEASE_PDB)` then breakpoints in executable works fine.
I am unsure if it's supposed to work that way, or only I have such problem or it's a real issue.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38641 (Git)
Build type: Release
Branch: main
Hash: 01d9e0c093ed35de24b45194040069a950132045
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

None

### Anything else?

I use Visual Studio 2022.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-09:
@oursland maybe you can help here?

oursland, 2024-09-09:
I'm really unfamiliar with Windows builds and the consequences of such an action, but it looks like the correct thing to do.

I have created a Draft PR #16399 that incorporates this change.  I have to again re-install a Windows VM to build and test the change, at which point it will be un-drafted and ready for review.

luzpaz, 2025-07-20:
cc @mdrost can you retest on latest 1.1-dev ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets