FreeCAD/FreeCAD #16370: Core: background process that watches to see how config files become corrupt overtime

luzpaz, 2024-09-08:
For years I've been seeing how the configs get corrupted and we don't really fully know why.  It would be beneficial if we ran some background process that just watched config history to see wth changes create these anomalies.

_Originally posted by @luzpaz in https://github.com/FreeCAD/FreeCAD/issues/16127#issuecomment-2318189930_

Roy-043, 2024-09-08:
There has been a recent PR (#13522) to address this. Have there been any examples of corrupted configs since then?

benj5378, 2024-09-08:
@luzpaz Can we gather examples of corruption to find similarities?

luzpaz, 2024-09-08:
Yes, hopefully that will be what this ticket achieves.

Roy-043, 2024-09-09:
Let's close this for now and just keep our eyes out for future CFG issues.

closed 2024-09-09 by Roy-043

luzpaz, 2024-09-14:
> We have a sizeable amount of issues triggered by "something" in the configs, which is really annoying.

https://github.com/FreeCAD/FreeCAD/issues/6106#issuecomment-2348801572

benj5378, 2024-10-09:
@luzpaz, before they reset their config file, we want them to make a backup of the config file that they can send us - in case the config shows to be the issue.