FreeCAD/FreeCAD #16371: Two shapes in a sketch can extrude only one body

HaydenHu, 2024-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-08:
Please provide a file.

HaydenHu, 2024-09-08:
> Please provide a file.
[DFN1006-3.zip](https://github.com/user-attachments/files/16922796/DFN1006-3.zip)

Roy-043, 2024-09-08:
You have to set the Body's "Allow Compound" property to true and then recompute the Pad. Please keep in mind that this is experimental. In V0.21 and earlier a Body must always be a single solid.

Next time please consider starting a forum topic before creating a GitHub issue. Closing this.

closed 2024-09-08 by Roy-043