FreeCAD/FreeCAD #16372: Open File Dialog (Property Editor) has layout problems

mnesarco, 2024-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When the user clicks the three dots button in a File property to select the external file, the Open File dialog is launched, but the dialog is initially in bad layout with header and footer hidden making it impossible to access its buttons.

https://github.com/user-attachments/assets/5566b6ae-ec85-4ca2-8519-6dc9a1e65e4b

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * DynamicData 2.62.0
  * MnesarcoUtils 0.2.6
  * sheetmetal 0.4.24
  * A2plus 0.4.67
  * fcscript 0.0.1
  * asm4 0.50.7
  * Curves 0.6.42
  * ProDarkThemePreferencePack 1.0.0
  * Marz 0.1.8
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mnesarco, 2024-09-08:
Apparently the problem is related to OpenTheme because classic does not have this problem.

mnesarco, 2024-09-08:
I am reporting to OpenTheme and closing here.

OpenTheme issue: https://github.com/obelisk79/OpenTheme/issues/122

mnesarco, 2024-11-22:
This issue is still present in 1.0, i am not sure if the problem is caused by core or opentheme

[image]

```
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * fcapi
  * Mnesarco_Utils 0.2.14
  * Marz 0.1.15
  * OpenTheme 2024.9.1

```

tarman3, 2024-11-22:
[image]

OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51

OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51

mnesarco, 2024-11-22:
Then it should be something related to OpenTheme

tarman3, 2024-11-22:
You can try disabling OpenTheme temporarily.

mnesarco, 2024-11-22:
I can confirm that the problem is only with OpenTheme.
Maybe @obelisk79 can provide some info.

luzpaz, 2025-03-13:
@obelisk79 ping

luzpaz, 2025-03-13:
Closing ticket as this is OpenThemes related and there is a downstream ticket dedicated to it, https://github.com/obelisk79/OpenTheme/issues/122

closed 2025-03-13 by luzpaz