FreeCAD/FreeCAD #16373: Draft: Polar array *without* fuse, reliable crash

theosib, 2024-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a prior polar array crash (https://github.com/FreeCAD/FreeCAD/issues/16026), but that happened when Fuse was enabled. The new one happens when fuse is NOT enabled. 

To get this crash to happen, select the "Sketch_on_Surface" and create a Draft polar array around the Z axis, count of 4. It will create the array, and then about a second later, it crashes.
[screw top.FCStd.zip](https://github.com/user-attachments/files/16922523/screw.top.FCStd.zip)

Also see the crash report from MacOS. 
[non-fuse-polar-crash.txt.zip](https://github.com/user-attachments/files/16922528/non-fuse-polar-crash.txt.zip)

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-09-08:
What's funny about this one is that enabling Fuse avoids the crash.

theosib, 2024-09-08:
Whoop. Spoke too soon. It crashed anyway. It just took about a minute. I was just rotating the view around and looking at it, and then it crashed.

Roy-043, 2024-09-08:
Work fine here:
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

theosib, 2024-09-08:
Well, for me, FreeCAD crashes instantly when I load the attached design. I'm running on Mac, BTW. Also, the original bug description includes a crash report.

[screw top.FCStd.zip](https://github.com/user-attachments/files/16923151/screw.top.FCStd.zip)

theosib, 2024-09-08:
I attached LLDB and ran FreeCAD from the command line. After I load my design, it's okay for about a minute, and then it crashes. I captured all the terminal output from that debug session. Please see the attachment.
[lldb-output.txt.zip](https://github.com/user-attachments/files/16923173/lldb-output.txt.zip)

theosib, 2024-09-08:
This crashes reliably in 38643, but it's fine in 38553, so this is a regression.

azuk, 2024-09-09:
Based on the stacktrace this is likely to be a duplicate of https://github.com/FreeCAD/FreeCAD/issues/16232. A fix for that was merged just a few hours ago, so test again when you can get a build that includes that fix.

theosib, 2024-09-17:
I'm on 38643, and it still crashes with a similar stack trace. Try the attachment.
[trash-can.FCStd.zip](https://github.com/user-attachments/files/17021262/trash-can.FCStd.zip)

theosib, 2024-09-17:
And here's another one that quickly crashes FreeCAD.
[trash can 2.FCStd.zip](https://github.com/user-attachments/files/17021329/trash.can.2.FCStd.zip)

theosib, 2024-09-17:
Ok, I tried them in 38768, and it works!

Roy-043, 2024-09-17:
Thanks for your feedback. Does this mean that this issue can be closed?

closed 2024-09-17 by luzpaz

theosib, 2024-09-17:
Looks like it should be closed. Thanks!