FreeCAD/FreeCAD #16382: Translation not reverted from Brazilian Portuguese to English

lucas-bueno777, 2024-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a fresh weekly build (Release Candidate 1, I think?), I started FreeCAD. The default language from the selector was European Portuguese, I then changed it to Brazilian Portuguese but decided to stick with English before proceeding. Notice that I didn't click OK when any Portuguese language was selected.

Now on the start screen, some text is still in Portuguese:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-08:
Does a restart fix it ?  

If so, looks related to: #6704

lucas-bueno777, 2024-09-09:
I just verified here and the answer is yes, restarting fixes the bug.

It would be just a minor inconvenience for the user trying the software at first, but after a restart everything is OK.

[image]

closed 2024-09-09 by luzpaz