FreeCAD/FreeCAD #16383: Assembly: accented words in Joint names doesn't show up correctly

MiqCG, 2024-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I suppose that Assembly WB has been partially translated into Catalan (European language) and the Revolute joint is translated "Revolució" with an accented O at the end.
In the attached picture you can see how new joint names appear as "Revoluci__" (while the old ones are kept as "Revolute", of course). The accented character is not displayed correctly. 
Don't know the implications of that but I haven't found any behavioural problem.

I guess this is related in some way to issue #12164

[image]

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * fasteners 0.5.27
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-08:
@pierreporte can you still reproduce this ?

pierreporte, 2024-09-13:
I can reproduce but it looks more complex than it appears. On the screenshot below, “éééééé” (1) is a renamed joint, and “Revoluci__” (2) was created as specified.

[image]

The issue is very similar to #12164 that was fixed by @PaddleStroke.

kwahoo2, 2024-09-19:
I can reproduce this in Polish:
[image]
```
OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * DesignSPHysics 0.7.0(15-09-2023)
  * Curves 0.6.47
  * freecad.gears 1.2.0
  * freecad-xr-workbench
```

PaddleStroke, 2024-10-08:
Fixed by https://github.com/FreeCAD/FreeCAD/pull/17108. 
It was actually exactly the same issue as in https://github.com/FreeCAD/FreeCAD/issues/12164 but just in a different place.