FreeCAD/FreeCAD #16384: New revisions of FreeCAD crash on delteing a feature of a particular file

MiqCG, 2024-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On revision 1.0.0RC1.38643 and 38553, FreeCAD crashes when trying to delete one of the three last items in the tree of the attached file. 
The file is a STEP imported from Grabcad and then "complemented" with Assembly3 objects. The object doesn't show up either in this revisions. Oddly enough it appears as is and doesn't crash on revisions 35966, 37543 and release 0.21.1.

[Bateria.zip](https://github.com/user-attachments/files/16923461/Bateria.zip)

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * fasteners 0.5.27
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-11:
Here are the backtraces.

<details><summary> Backtraces</summary>

<details><summary>gdb</summary>

```cpp
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
App::LinkBaseExtension::getLinkCopyOnChangeValue (this=0x0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/Link.h:246
246         BOOST_PP_SEQ_FOR_EACH(LINK_PROP_GET,_,LINK_PARAMS)
(gdb) bt
#0  App::LinkBaseExtension::getLinkCopyOnChangeValue (this=0x0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/Link.h:246
#1  0x00007ffff535f919 in App::LinkBaseExtension::isLinkMutated (this=this@entry=0x0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/App/Link.cpp:2233
#2  0x00007ffff7ce5c74 in Gui::ViewProviderLink::onDelete (this=this@entry=0x55555a367bc0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/ViewProviderLink.cpp:2387
#3  0x00007ffff7cfb1ae in Gui::ViewProviderFeaturePythonT<Gui::ViewProviderLink>::onDelete (this=0x55555a367bc0, sub=std::vector of length 0, capacity 0)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/ViewProviderFeaturePython.h:306
#4  0x00007ffff7962174 in StdCmdDelete::activated (this=<optimized out>, iMsg=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/CommandDoc.cpp:1409
#5  0x00007ffff7952d4c in Gui::Command::_invoke (this=this@entry=0x555555c4a980, id=id@entry=0, 
    disablelog=<optimized out>) at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Command.cpp:444
#6  0x00007ffff79533bc in Gui::Command::invoke (this=0x555555c4a980, i=i@entry=0, 
    trigger=trigger@entry=Gui::Command::TriggerAction)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Command.cpp:408
#7  0x00007ffff793e361 in Gui::Action::onActivated (this=<optimized out>)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Action.cpp:111
#8  0x00007ffff7947e9a in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(void (Gui::Action::*)(), Gui::Action*, void**) (f=<optimized out>, o=<optimized out>, arg=<optimized out>) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152
#9  0x00007ffff7947ed8 in QtPrivate::FunctionPointer<void (Gui::Action::*)()>::call<QtPrivate::List<>, void>(void (Gui::Action::*)(), Gui::Action*, void**) (arg=<optimized out>, o=<optimized out>, f=<optimized out>)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185
#10 QtPrivate::QSlotObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=<optimized out>, this_=<optimized out>, r=<optimized out>, a=<optimized out>, ret=<optimized out>)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:418
#11 0x00007ffff2b12e16 in QtPrivate::QSlotObjectBase::call (a=0x7fffffffc520, r=0x555557a9d240, 
    this=0x555557a9d150) at ../../include/QtCore/../../src/corelib/kernel/qobjectdefs_impl.h:398
#12 doActivate<false> (sender=0x55555784d170, signal_index=4, argv=0x7fffffffc520) at kernel/qobject.cpp:3925
#13 0x00007ffff2b0b697 in QMetaObject::activate (sender=sender@entry=0x55555784d170, 
    m=m@entry=0x7ffff3c8bc80 <QAction::staticMetaObject>, local_signal_index=local_signal_index@entry=1, argv=argv@entry=0x7fffffffc520) at kernel/qobject.cpp:3985
#14 0x00007ffff3764f94 in QAction::triggered (this=this@entry=0x55555784d170, _t1=<optimized out>)
    at .moc/moc_qaction.cpp:376
#15 0x00007ffff3767eab in QAction::activate (this=0x55555784d170, event=<optimized out>)
    at kernel/qaction.cpp:1161
#16 0x00007ffff799d3ec in Gui::ShortcutManager::onTimer (this=this@entry=0x555555ed1330)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/ShortcutManager.cpp:428
#17 0x00007ffff799e07e in Gui::ShortcutManager::checkShortcut (this=0x555555ed1330, o=<optimized out>, 
    key=...) at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/ShortcutManager.cpp:239
#18 0x00007ffff799e88b in Gui::ShortcutManager::eventFilter (this=0x555555ed1330, o=<optimized out>, 
    ev=<optimized out>) at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/ShortcutManager.cpp:284
#19 0x00007ffff2ad7d5e in QCoreApplicationPrivate::sendThroughApplicationEventFilters (
    this=this@entry=0x5555558c3210, receiver=receiver@entry=0x55555784d170, event=event@entry=0x7fffffffcae0)
    at kernel/qcoreapplication.cpp:1172
#20 0x00007ffff376bd80 in QApplicationPrivate::notify_helper (this=this@entry=0x5555558c3210, 
    receiver=0x55555784d170, e=0x7fffffffcae0, e@entry=0xff7d5857eb72a800) at kernel/qapplication.cpp:3611
#21 0x00007ffff3773547 in QApplication::notify (this=<optimized out>, receiver=<optimized out>, 
    e=0xff7d5857eb72a800, e@entry=0x7fffffffcae0) at kernel/qapplication.cpp:3386
#22 0x00007ffff7915c5c in Gui::GUIApplication::notify (this=0x7fffffffd410, receiver=0x55555784d170, 
    event=0x7fffffffcae0) at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/GuiApplication.cpp:83
#23 0x00007ffff2ad8118 in QCoreApplication::notifyInternal2 (receiver=0x55555784d170, event=0x7fffffffcae0)
    at kernel/qcoreapplication.cpp:1064
#24 0x00007ffff2ad8332 in QCoreApplication::sendEvent (receiver=<optimized out>, event=<optimized out>)
    at kernel/qcoreapplication.cpp:1462
#25 0x00007ffff2f77b3c in QShortcutMap::dispatchEvent (this=<optimized out>, e=<optimized out>)
    at kernel/qshortcutmap.cpp:675
#26 0x00007ffff2f787b3 in QShortcutMap::tryShortcut (this=this@entry=0x5555558c32d8, 
    e=e@entry=0x7fffffffcbb0) at kernel/qshortcutmap.cpp:343
[... cut ...]
    frame #37: 0x00007ffff202a28b libc.so.6`__libc_start_main_impl(main=(FreeCAD`main), argc=1, argv=0x00007fffffffda08, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffd9f8) at libc-start.c:360:3
    frame #38: 0x0000555555570995 FreeCAD`_start + 37
(lldb)
```

</details>

</details>

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38694 (Git)
Build type: Debug
Branch: master
Hash: 614a14fe87af6ea0a978ee6312b298e4033f1839
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: //
```

chennes, 2024-10-16:
I can't run a recompute on this file, it looks like it has some kind of old element map in it? Was this originally an LS3 file?

MiqCG, 2024-10-16:
I cannot remember precisely with which version it was created. All I know for sure is that at some point it was definitely involved in an Assembly 3 file (it has already the assembly elements).

If I open the file with 0.21.1 and delete the Assembly3 elements (outside the Body), then the file can be opened and recognized in RC1 and RC2.

maxwxyz, 2024-10-17:
I'll remove the blocker tag then

closed 2025-11-21 by github-actions[bot]