FreeCAD/FreeCAD #16386: BIM: error when trying to move objects (M,V)

gianlucafiore, 2024-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I try to move the selected slabs 100mm upwards, it moves them to one side, and nothing on the Z axis

TEST FILE: [edificio nuevo 2.zip](https://github.com/user-attachments/files/16923607/edificio.nuevo.2.zip)

[image]
[image]
[image]

Finally result
[image]

1- Select the 6 slabs
2- Apply the move command (M,V)
3- Restrict the z axis to enter 100mm
4- At the end, the slabs all moved to the right a random distance

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gianlucafiore, 2024-09-08:
[image]
I just realized that it is not taking the distance entered in the forced z axis

Roy-043, 2024-09-09:
What you are doing is very strange. You are constraining to Z while the current workingplane is Top, and then not clicking points in the 3D view.

To achieve what you want the obvious choice would be to enter 0,0,0 for the first point and 0,0,100 for the second. But please explain your steps, and your logic, in more detail.

gianlucafiore, 2024-09-09:
[image]
It seems to me that what I am suggesting makes sense, in fact I changed the work plane, I am still stuck to the Z axis, and the problem persists. The total distance that appears in the coordinates should be those of the Z axis. I don't know what the problem is with what I am suggesting.

gianlucafiore, 2024-09-09:
If I enter the measurement directly without removing the Z axis forcing, it is entered through the X axis and moves the object to the side

[image]
[image]

I don't think it's the most intuitive thing

yorikvanhavre, 2024-09-10:
I think the problem is that the working plane is set and you are snapping to it... In any case, this works for me:

- the working plane is set to top (or whatever else, the point is, it's not auto)
- select an object
- start move command (M,V)
- pick a point with the mouse
- press Z to restrict to z
- enter 200cm and press enter

The object is correctly moved 200cm high

Roy-043, 2024-09-10:
I think one of the key factors is where the cursor is when you press Z. You should press that key while the mouse is still over the 3D view.

azuk, 2024-09-10:
> I think one of the key factors is where the cursor is when you press Z. You should press that key while the mouse is still over the 3D view.

Regardless of where my cursor is when I press Z, whether in 3D view (outside of any object or over an object) or over Move dialog, whatever is in X and Y fields of Move dialog remains in those now disabled fields. However, after pressing Z, if I move cursor in 3D view, X and Y fields are set to 0.

I think it would be better if X and Y fields (in this case of restricting to Z) were set to 0 immediately when Z is pressed so that there is no need to move the mouse. If you have something in X and Y, that's going to affect Length. And if you modify Length, it will change non-zero values in those disabled X and Y fields.

It also seems that you can restrict to X, Y or Z when picking the first point. Is this intentional? Anyway, in this case after restricting to for example Z and moving cursor in 3D view, X and Y fields are NOT set to 0, so behaviour is different than with the second point.

>     * start move command (M,V)
>     * pick a point with the mouse
>     * press Z to restrict to z

This doesn't actually work for me. I need to move mouse cursor between picking the first point and pressing Z, otherwise pressing Z doesn't do anything.

```
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: recent FreeCAD-1-0 branch build, see hash
Build type: Debug
Branch: main
Hash: e3ac0e4d092b8a44a7e045f18f5c9af2e07ba568
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1.dev
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
Installed mods: 
  * Help 1.0.3
  * Render 2024.8.4
  * Curves 0.6.47
  * Assembly4 0.50.15
  * fasteners 0.5.28
  * Assembly3 0.12.2
  * CurvedShapes 1.0.9
```

yorikvanhavre, 2024-09-11:
Indeed I can confirm that, the problem is when the X and Y fields contain a value when Z is pressed. And I also think the two other fields should be set to 0 when a restricting X, Y or Z key is pressed. That's what the Shift key does already, actually.

PR #16434 (2024-09-13): Draft: Fixed restrained movement
Fixes #16386

This should go to 1.0
commit f09dfa72 (2024-09-13): Draft: Fixed restrained movement - Fixes #16386
commit 14ff4cb8 (2024-09-13): Draft: Fixed restrained movement - Fixes #16386
closed 2024-09-13 by chennes