FreeCAD/FreeCAD #16387: Start: Typo in start page

luzpaz, 2024-09-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@concretedog shared an image of a running RC1.0 on Raspberry Pi4 and found a typo

[image]

[image]

But searching the source, it isn't present in the source
https://github.com/FreeCAD/FreeCAD/blob/7ed1e9380acccd686a0e3573f883ac793aec3299/src/Mod/Start/Gui/GeneralSettingsWidget.cpp#L213

### Full version info

```shell
TBD
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mdrost, 2024-09-08:
Probably the screen is not big enough. I "reproduced" the issue by resizing FreeCAD window.

luzpaz, 2024-09-08:
So weird that it truncates the 'm' to look like an 'n'.  In that case. I'll close ticket. Thought I'd open it JIC.

closed 2024-09-08 by luzpaz