FreeCAD/FreeCAD #16392: OpenBSD: Segfault when using the Transformation Tool

thyssentishman, 2024-09-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm working on a [FreeCAD port for OpenBSD](https://forum.freecad.org/viewtopic.php?t=80792) and I'm building from HEAD (fe40748). Whenever I use the transformation tool (Right click -> Transform) on a body and then then click 'Ok' or 'Cancel' on the Tasks panel to finish using the tool, I get a segmentation fault.

Steps to reproduce:
1. New document
2. Create Part->Cube
3. Right click on 'Cube' -> Transform
4. Rotate/Move the cube
5. Click 'Ok' or 'Cancel'

Without step 4. there is no segmentation fault.

<details>
<summary>Backtrace</summary><pre>
Reading symbols from /usr/local/lib/freecad/bin/FreeCAD...
[New process 422390]
[New process 572552]
[New process 263223]
[New process 463511]
[New process 566821]
[New process 609552]
[New process 228947]
[New process 138807]
[New process 580413]
[New process 408596]
[New process 437266]
[New process 198504]
[New process 147058]
[New process 367963]
[New process 382344]
[New process 593566]
[New process 377014]
[New process 308958]
[New process 416647]
[New process 387749]
[New process 591681]
[New process 611945]
[New process 582674]
[New process 131123]
[New process 479808]
[New process 598771]
[New process 314990]
[New process 614183]
[New process 484580]
[New process 227272]
[New process 505179]
[New process 195940]
[New process 218461]
[New process 104472]
[New process 372795]
Core was generated by `FreeCAD'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000087f45a3b0f0 in SoCallbackAction::~SoCallbackAction() () from /usr/local/lib/libCoin.so.0.0
[Current thread is 1 (process 422390)]
(gdb) bt
#0  0x0000087f45a3b0f0 in SoCallbackAction::~SoCallbackAction() () from /usr/local/lib/libCoin.so.0.0
#1  0x0000087f45a3b293 in SoCallbackAction::~SoCallbackAction() () from /usr/local/lib/libCoin.so.0.0
#2  0x0000087f45aaf47f in SoDragger::~SoDragger() () from /usr/local/lib/libCoin.so.0.0
#3  0x0000087f2e4914e3 in Gui::TDragger::~TDragger() () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#4  0x0000087f45c39efc in SoBase::destroy() () from /usr/local/lib/libCoin.so.0.0
#5  0x0000087f45c2c69a in SoBaseList::truncate(int) () from /usr/local/lib/libCoin.so.0.0
#6  0x0000087f45c3e805 in SoChildList::~SoChildList() () from /usr/local/lib/libCoin.so.0.0
#7  0x0000087f45c97ebe in SoGroup::~SoGroup() () from /usr/local/lib/libCoin.so.0.0
#8  0x0000087f45cb5f03 in SoSeparator::~SoSeparator() () from /usr/local/lib/libCoin.so.0.0
#9  0x0000087f45c39efc in SoBase::destroy() () from /usr/local/lib/libCoin.so.0.0
[... 57 lines cut ...]
#67 0x0000087f2e0aebe4 in Gui::Application::runApplication() () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#68 0x0000087d0524e6bc in main ()
(gdb) quit

</pre>
</details>

### Full version info

```shell
OS: OpenBSD 7.6
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.Unknown
Build type: Release
Python 3.11.9, Qt 6.6.3, Coin 4.0.2, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-09:
Can you reproduce the crash with _FreeCAD_ compiled against _Coin 4.0.0_?

thyssentishman, 2024-09-09:
> Can you reproduce the crash with _FreeCAD_ compiled against _Coin 4.0.0_?

I'm having a hard time trying to compile Coin 4.0.0 on OpenBSD. I was able to test with Coin 4.0.1 and the issue is also there, FreeCAD segfaults.

thyssentishman, 2024-09-11:
I recompiled Coin 4.0.3 and FreeCAD with debug info. Below is the backtrace I got.

<details>
<summary>Backtrace</summary><pre>
Reading symbols from /usr/local/lib/freecad/bin/FreeCAD...
[New process 226691]
[New process 322723]
[New process 485712]
[New process 556225]
[New process 263247]
[New process 552850]
[New process 232737]
[New process 467664]
[New process 503348]
[New process 273224]
[New process 513382]
[New process 505209]
[New process 248389]
[New process 164977]
[New process 236367]
[New process 167420]
[New process 363244]
[New process 528513]
[New process 623304]
[New process 319956]
[New process 611297]
[New process 132500]
[New process 275632]
[New process 583068]
[New process 347336]
[New process 443627]
[New process 319055]
[New process 572133]
[New process 580104]
[New process 349824]
[New process 543980]
[New process 598738]
[New process 614162]
[New process 259422]
[New process 529042]
[New process 168898]
[New process 208366]
[New process 315616]
[New process 571862]
[New process 104953]
[New process 259266]
[New process 337345]
[New process 333827]
[New process 479795]
Core was generated by `FreeCAD'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  SoCallbackData::deleteAll (this=<optimized out>) at /usr/ports/pobj/coin-4.0.3/coin/src/actions/SoCallbackAction.cpp:242
242           nextptr = cbdata->next;
[Current thread is 1 (process 226691)]
(gdb) bt
#0  SoCallbackData::deleteAll (this=<optimized out>) at /usr/ports/pobj/coin-4.0.3/coin/src/actions/SoCallbackAction.cpp:242
#1  delete_list_elements (cl=...) at /usr/ports/pobj/coin-4.0.3/coin/src/actions/SoCallbackAction.cpp:465
#2  SoCallbackAction::~SoCallbackAction (this=0x7f4838a2700) at /usr/ports/pobj/coin-4.0.3/coin/src/actions/SoCallbackAction.cpp:474
#3  0x000007f4d5e392c3 in SoCallbackAction::~SoCallbackAction (this=0x0) at /usr/ports/pobj/coin-4.0.3/coin/src/actions/SoCallbackAction.cpp:472
#4  0x000007f4d5ead4df in SoDragger::~SoDragger (this=0x7f511819a00) at /usr/ports/pobj/coin-4.0.3/coin/src/draggers/SoDragger.cpp:396
#5  0x000007f48dd5e5a3 in Gui::TDragger::~TDragger (this=0x0) at /usr/ports/pobj/freecad-1.0/FreeCAD-fe40748c10503eed367840239db3e0f81873e359/src/Gui/SoFCCSysDragger.cpp:141
#6  0x000007f4d6037f8c in SoBase::destroy (this=0x7f511819a00) at /usr/ports/pobj/coin-4.0.3/coin/src/misc/SoBase.cpp:334
#7  0x000007f4d602a72a in SoBaseList::truncate (this=0x7f4b83020e0, length=0) at /usr/ports/pobj/coin-4.0.3/coin/src/lists/SoBaseList.cpp:181
#8  0x000007f4d603c895 in SoChildList::~SoChildList (this=0x0) at /usr/ports/pobj/coin-4.0.3/coin/src/misc/SoChildList.cpp:94
#9  0x000007f4d6095f4e in SoGroup::~SoGroup (this=0x7f4a6e02d80) at /usr/ports/pobj/coin-4.0.3/coin/src/nodes/SoGroup.cpp:281
#10 0x000007f4d60b3fa3 in SoSeparator::~SoSeparator (this=0x0) at /usr/ports/pobj/coin-4.0.3/coin/src/nodes/SoSeparator.cpp:465
#11 0x000007f4d6037f8c in SoBase::destroy (this=0x7f4a6e02d80) at /usr/ports/pobj/coin-4.0.3/coin/src/misc/SoBase.cpp:334
#12 0x000007f4d602a72a in SoBaseList::truncate (this=0x7f510286930, length=0) at /usr/ports/pobj/coin-4.0.3/coin/src/lists/SoBaseList.cpp:181
#13 0x000007f4d603c895 in SoChildList::~SoChildList (this=0x0) at /usr/ports/pobj/coin-4.0.3/coin/src/misc/SoChildList.cpp:94
#14 0x000007f4d6095f4e in SoGroup::~SoGroup (this=0x7f4ed1c2000) at /usr/ports/pobj/coin-4.0.3/coin/src/nodes/SoGroup.cpp:281
[... 42 lines cut ...]
#57 0x000007f4fa660414 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#58 0x000007f48da5ce9d in Gui::GUIApplication::notify (this=0x6fc2f46e8890, receiver=0x7f48eadd300, event=0x6fc2f46e80b0) at /usr/ports/pobj/freecad-1.0/FreeCAD-fe40748c10503eed367840239db3e0f81873e359/src/Gui/GuiApplication.cpp:83
#59 0x000007f516716e76 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/libQt6Core.so.4.1
#60 0x000007f4e2acdd4f in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/local/lib/libQt6Gui.so.6.0
#61 0x000007f4e2b37b5b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Gui.so.6.0
#62 0x000007f566253fc3 in xcbSourceDispatch(_GSource*, int (*)(void*), void*) () from /usr/local/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.3.0
#63 0x000007f51879511f in g_main_context_dispatch_unlocked () from /usr/local/lib/libglib-2.0.so.4201.12
#64 0x000007f51879550d in g_main_context_iterate_unlocked () from /usr/local/lib/libglib-2.0.so.4201.12
#65 0x000007f5187955bb in g_main_context_iteration () from /usr/local/lib/libglib-2.0.so.4201.12
#66 0x000007f51699f7e2 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Core.so.4.1
#67 0x000007f516721385 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Core.so.4.1
#68 0x000007f516717613 in QCoreApplication::exec() () from /usr/local/lib/libQt6Core.so.4.1
#69 0x000007f48d97bbd4 in (anonymous namespace)::tryRunEventLoop (mainApp=...) at /usr/ports/pobj/freecad-1.0/FreeCAD-fe40748c10503eed367840239db3e0f81873e359/src/Gui/Application.cpp:2165
#70 (anonymous namespace)::runEventLoop (mainApp=...) at /usr/ports/pobj/freecad-1.0/FreeCAD-fe40748c10503eed367840239db3e0f81873e359/src/Gui/Application.cpp:2185
#71 Gui::Application::runApplication () at /usr/ports/pobj/freecad-1.0/FreeCAD-fe40748c10503eed367840239db3e0f81873e359/src/Gui/Application.cpp:2262
[... cut ...] bj/freecad-1.0/FreeCAD-fe40748c10503eed367840239db3e0f81873e359/src/Main/MainGui.cpp:301
(gdb) quit
</pre>
</details>

The crash happens [here](https://github.com/coin3d/coin/blob/2e7e4d041a9bce29ac288ab1aa40d4ed5a00bac7/src/actions/SoCallbackAction.cpp#L242) when de-referencing the `cbdata` pointer:

https://github.com/coin3d/coin/blob/2e7e4d041a9bce29ac288ab1aa40d4ed5a00bac7/src/actions/SoCallbackAction.cpp#L238-L246

Not sure what happens here. The pointer should be guarded by the while loop.

closed 2024-09-16 by thyssentishman