FreeCAD/FreeCAD #16393: FreeCAD from daily PPA crashes when creating a new document on Ubuntu 24.04

furgo16, 2024-09-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After upgrading to Ubuntu 24.04 and to the corresponding FreeCAD packages from the daily builds, FreeCAD does not seem to be fully usable. In addition to frequent freezes, I can reproduce this crash

Steps to reproduce:

1. Start FreeCAD (freecad-daily) from the daily PPA
2. Notice a few things: 1) the launcher icon is the old FreeCAD logo, 2) FreeCAD seems to freeze the whole system at short intervals 3) the report view shows `QOpenGLFunctions created with non-current context` and `(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()`
3. Click on the button to create a new document
4. FreeCAD crashes. Output on the terminal:

```
QSocketNotifier: Can only be used with threads started with QThread
QOpenGLFunctions created with non-current context
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()
freecad-daily: ./src/glue/gl.cpp:2234: const cc_glglue* cc_glglue_instance(int): Assertion `current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"' failed.
```

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38644 (Git)
Build type: Release
Branch: main
Hash: 7ed1e9380acccd686a0e3573f883ac793aec3299
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: C/Default (C)
Installed mods: 
  * OpenTheme 2024.7.3
  * WikiPropertiesGenerator.FCMacro
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-09:
> Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.

is this relevant ?

wsteffe, 2024-09-09:
I had the same problem after upgrading ubuntu to 24.04.1. I reported it at https://github.com/realthunder/FreeCAD/issues/1031 but then I have also seen that this problem goes away executing:

`QT_QPA_PLATFORM=xcb  FreeCAD`

instead of:
 
`FreeCAD`

furgo16, 2024-09-09:
That works indeed!

```
QT_QPA_PLATFORM=xcb freecad-daily
```

furgo16, 2024-09-09:
>>Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.
>>
> is this relevant ?

I do not think so. The message itself is a bit vague, but my guess is that Ubuntu 24.04 does not ship an "old" Mesa GL version.

furgo16, 2024-09-10:
Crash also happens executing the binary on a local build. The workaround works. If there was a way for the PPA to add the `QT_QPA_PLATFORM=xcb` line to the package (e.g. on the .desktop file), it would be great to have FreCAD working out of the box.

Related: https://github.com/FreeCAD/FreeCAD/issues/13144

```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Debug
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
Installed mods: 
  * OpenTheme 2024.7.3
  * WikiPropertiesGenerator.FCMacro
```

furgo16, 2024-10-01:
This seems not to be specific to the PPA package. Specifying the `QT_QPA_PLATFORM` environment variable is required to avoid the crash on local builds as well.

closed 2025-08-03 by github-actions[bot]