FreeCAD/FreeCAD #16398: PartDesign: sub-object(s) shape binder fails to to project correctly the selected object face

MineralAB, 2024-09-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sub-object(s) shape binder fails to project correctly the selected object face.

Here some screenshots:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I also attach the file without any Sub-object(s) shape binder created yet:
[FCTest2.zip](https://github.com/user-attachments/files/16934742/FCTest2.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.67
  * Render 2024.8.4
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-09:
@MineralAB is this the same issue in the stable version?
@bgbsww is this the same as: https://github.com/FreeCAD/FreeCAD/issues/14379 ?

MineralAB, 2024-09-09:
@maxwxyz i think the problem was deeper than i thought.
After the post i closed FreeCAD, now i reopened the same file and this is what i see: 
[image]
The binder seems to work, but somehow the body itself broke and the render of the body didn't update.

luzpaz, 2024-09-09:
@howetuft do you mind participating since this involves Render WB ? TIA

howetuft, 2024-09-10:
Hello @luzpaz,
Thank you, but I can't find how this issue involves Render (I think it's rather related to internal viewport).

closed 2025-07-20 by maxwxyz