FreeCAD/FreeCAD #16400: [TechDraw] QGIViewDimension unreadable technical debt

benj5378, 2024-09-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

TechDrawGui::QGIViewDimension are 1200 lines long and unreadable. This makes changes (that I was trying to make) very difficult.

Here are two main issues:
QGIViewDimension doesn't use object-oriented programming, it throws around 6-11 arguments for creating a dimension line, like member variables doesn't exist.

QGIViewDimension tries to be 10 different dimension types at once. This means that there are 10 different functions for each dimension types, completely independent of each other. There is no reason for having all this in a single object; it should really have been 10 different objects; that would significantly reduce the logic trying to differentiate the different types and improve readability only having related functions in the same file/object.

### Full version info

```shell
-
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-09-09:
What do you think, @WandererFan? Wouldn't it be a huge improvement splitting the object into multiple objects?

mdrost, 2024-09-09:
> QGIViewDimension (...) throws around 6-11 arguments for creating a dimension line, like member variables doesn't exist.

I would say it is a good thing. IMHO if the scope of some variables is limited and they are not needed for the lifetime of an object then they should not be member variables of that object.

kadet1090, 2024-09-09:
This class is not so bad, it has a lot of stuff in it but it is quite decent in terms of what else you can find in FreeCAD. As for the number of arguments - they seem quite reasonable looking at how they are used. Making this component stateful does not seem right for me also. Stateful classes are much harder (impossible really, as you can never be sure about the full state) to reason about. 

That said, many arguments are repeated over and over again and so creating proper structures / classes to combine them together could be beneficial.

benj5378, 2024-09-09:
@kadet1090 @mdrost What about splitting the class into multiple classes?

mdrost, 2024-09-10:
I am very new to this project, so I have no strong opinion regarding splitting the class.

luzpaz, 2025-05-10:
Just checking if this is still relevant ?

WandererFan, 2025-05-12:
QGIViewDimension is still an over complex mess.  Very hard to work on.

closed 2025-10-04 by github-actions[bot]

FEA-eng, 2026-08-15:
Since it's about refactoring, I assume that it's still relevant.