FreeCAD/FreeCAD #16406: Seg Fault crash when moving mouse over main document.

Connor9220, 2024-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have an existing project.  When I mouse over the main document window, FreeCAD crashes with seg fault.  I've tracked it down to  using the Draft Array on a body.  I turn off viability and it doesn't crash any more.   This looks to be a regression.   Bug appeared after weekly build 38622 and still present in 38643.  Works in 38555
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7ffbc2e5b050]
#1  0x7ffbc63bfed3 in Gui::LinkView::linkGetElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x543
#2  0x7ffbc63c0423 in Gui::ViewProviderLink::getElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x113
#3  0x7ffbc623a9d2 in Gui::SoFCUnifiedSelection::getPickedList(SoHandleEventAction*, bool) const from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x572
#4  0x7ffbc623f797 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x1e7
#5  0x7ffbc26c5e38 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libCoin.so.80+0x68
#6  0x7ffbc287fcca in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libCoin.so.80+0x15a
#7  0x7ffbc28cfce8 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libCoin.so.80+0x58
#8  0x7ffbc28eaf8a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libCoin.so.80+0x2a
#9  0x7ffbc26c5e38 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libCoin.so.80+0x68
#10  0x7ffbc26cfa2c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libCoin.so.80+0xac
#11  0x7ffbc26c6b1a in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libCoin.so.80+0x2ea
#12  0x7ffbc289bdd2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libCoin.so.80+0x72
#13  0x7ffbc62edfee in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x1e
#14  0x7ffbc62f9588 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x58
#15  0x7ffbc6306230 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x40
#16  0x7ffbc63129f7 in Gui::GestureNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x2e7
#17  0x7ffbc63064c5 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x65
#18  0x7ffbc632e6bb in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x6b
#19  0x7ffbc62e491d in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x1fd
#20  0x7ffbc32b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Core.so.5+0x7b
#21  0x7ffbc417039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#22  0x7ffbc41778f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#23  0x7ffbc5f3e499 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x99
#24  0x7ffbc32b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Core.so.5+0x108
#25  0x7ffbc41763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#26  /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7ffbc41c6e6f]
#27  /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7ffbc41ca0c9]
#28  0x7ffbc41703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#29  0x7ffbc5f3e499 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x99
#30  0x7ffbc32b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Core.so.5+0x108
#31  0x7ffbc39353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Gui.so.5+0x84c
#32  0x7ffbc390b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Gui.so.5+0xac
#33  /tmp/.mount_FreeCAK7CzsU/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7ffbbdcdebea]
#34  /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7ffbc1b10ad7]
#35  /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7ffbc1b13ac7]
#36  /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7ffbc1b140ce]
#37  0x7ffbc330aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Core.so.5+0x66
#38  0x7ffbc32b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Core.so.5+0x11b
#39  0x7ffbc32bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libQt5Core.so.5+0x82
#40  0x7ffbc5e9f943 in Gui::Application::runApplication() from /tmp/.mount_FreeCAK7CzsU/usr/bin/../lib/libFreeCADGui.so+0x983
#41  /tmp/.mount_FreeCAK7CzsU/usr/bin/freecad(+0x7640) [0x55c1f1366640]
#42  /lib/x86_64-linux-gnu/libc.so.6(+0x2724a) [0x7ffbc2e4624a]
#43  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7ffbc2e46305]
#44  /tmp/.mount_FreeCAK7CzsU/usr/bin/freecad(+0x79a9) [0x55c1f13669a9]

``` 

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ThreadProfile.backup1725502069.7449253 1.90.0 (Disabled)
  * QuickMeasure 2022.10.28
  * dodo 1.0.1
  * btl.backup1725502066.664948 0.9.9 (Disabled)
  * btl.backup1720304573.9159045 0.9.9 (Disabled)
  * A2plus.backup1725502065.0763218 0.4.65 (Disabled)
  * btl 0.9.9 (Disabled)
  * OpenDark 2024.3.13
  * A2plus 0.4.67
  * btl.backup1717184435.3801382 0.9.9 (Disabled)
  * lattice2 1.0.0
  * ThreadProfile 1.92.0
  * fasteners.backup1725502068.289526 0.5.24 (Disabled)
  * fasteners 0.5.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-09-10:
Perhaps another duplicate of https://github.com/FreeCAD/FreeCAD/issues/16232?

Connor9220, 2024-09-10:
> Perhaps another duplicate of #16232?

Going back and reading the other bug (Which I didn't find for some reason)  This is most likely the same issue. I'll test as soon as the next weekly/RC is released.

benj5378, 2024-10-05:
@Connor9220, can you check on new version?

maxwxyz, 2024-10-16:
@Connor9220 please test, otherwise I'll close this issue.

luzpaz, 2024-10-16:
Closing. If reproducible, please notate ticket and we'll re-open.

closed 2024-10-16 by luzpaz

Connor9220, 2024-10-17:
Sorry, didn't see this in time. Works in RC2