FreeCAD/FreeCAD #16407: Headless .py scripting - import PartDesignGui fails

c3kkos, 2024-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using external .py scripting, trying to use a PartDesign feature.

### Full version info

```shell
Using vanilla Ubuntu Server 22.04 LTS 64bit on a virtual machine

FreeCAD version:

>>> App.Version()
['1', '0', '0', '38684 (Git)', 'https://github.com/FreeCAD/FreeCAD main', '2024/09/09 16:35:22', 'main', '92e31fa7ea974fe48c372f6749916d68f42dcf95']

This is my .py script:

FREECADPATH= '/usr/lib/freecad-daily-python3/lib'
import sys
sys.path.append(FREECADPATH)
print ('1 this prints')
from pivy import coin
print ('2 this prints')
import FreeCAD as App
import FreeCADGui as Gui
print ('3 this prints')
Gui.setupWithoutGUI()
print ('4 this prints')
import Part
import PartDesignGui
print ('5 THIS DONT PRINT')

...

there is no need to attach further code here, everything below "import PartDesignGui" simply doesnt get executed, hence i've figured out something crashes out without visible errors/warning (no output at all, i've checked dmesg too)

Why importing Gui?
Because a "PartDesign" alone module seems not importable, so the workaround is shown above:
importing Gui and using the appropriate method Gui.setupWithoutGUI(), to be able to import PartDesignGui

Why not using the Part module, which is importable without issues?
Because i want to use simple additive and subtractive primitives like boxes and cylinders, and PartDesign offers a WAY much clearer and simple approach in doing so.

Actually, there is a .so file in lib dir, it is named _PartDesign.so
I can't understand that underscore at the bebinning, though i'm sure there's a good reason
Anyhow, import _PartDesign simply don't work (is it a submodule?)

Goal:

This work aims to automate a process that calls the .py script to render some simple 3d pieces to be used in another software or to be served via web
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

c3kkos, 2024-09-10:
using freecadcmd-daily, importing PartDesignGui gives SIGSEGV, here's the log:

`Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7feea29a5520]
#1  0x7fee94f0e1f7 in QKeySequence::keyBindings(QKeySequence::StandardKey) from                                                                                                                        /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x27
#2  0x7fee94f0e277 in QKeySequence::QKeySequence(QKeySequence::StandardKey) from                                                                                                                        /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x27
#3  0x7fee96864a72 in Gui::Command::keySequenceToAccel(int) const from /usr/lib/                                                                                                                       freecad-daily-python3/lib/libFreeCADGui.so+0x92
#4  0x7fee934bc27e in CmdSketcherCopyClipboard::CmdSketcherCopyClipboard() from                                                                                                                        /usr/lib/freecad-daily-python3/lib/SketcherGui.so+0x6e
#5  0x7fee934cbef2 in CreateSketcherCommandsConstraintAccel() from /usr/lib/free                                                                                                                       cad-daily-python3/lib/SketcherGui.so+0x332
#6  /usr/lib/freecad-daily-python3/lib/SketcherGui.so(PyInit_SketcherGui+0x39e)                                                                                                                        [0x7fee933cc71e]
#7  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1e9071) [0x7feea2fe4071]
#8  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1287b3) [0x7feea2f237b3]
#9  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b16)                                                                                                                        [0x7feea2e71776]
#10  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7feea2fbd3af]
#11  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68)                                                                                                                        [0x7feea2e769c8]
#12  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7feea2fbd3af]
#13  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x829e)                                                                                                                        [0x7feea2e74efe]
#14  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7feea2fbd3af]
#15  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69de)                                                                                                                        [0x7feea2e7363e]
#16  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7feea2fbd3af]
#17  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69de)                                                                                                                        [0x7feea2e7363e]
#18  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7feea2fbd3af]
#19  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69de)                                                                                                                        [0x7feea2e7363e]
#20  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7feea2fbd3af]
#21  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0xe13f4) [0x7feea2edc3f4]
#22  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_CallMethodIdObjArgs+0x                                                                                                                       ff) [0x7feea2edc75f]
#23  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyImport_ImportModuleLevelObject                                                                                                                       +0x2a4) [0x7feea2fe62a4]
#24  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1be2fe) [0x7feea2fb92fe]
#25  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128023) [0x7feea2f23023]
#26  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_Call+0x5c) [0x7feea2ed                                                                                                                       cfec]
#27  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b16)                                                                                                                        [0x7feea2e71776]
#28  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7feea2fbd3af]
#29  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x5bf7)                                                                                                                        [0x7feea2e72857]
#30  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7feea2fbd3af]
#31  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x7feea2f                                                                                                                       b83de]
#32  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4cd) [0x7feea30074cd]
#33  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x7feea                                                                                                                       30084f9]
[... cut ...] ]
#60  0x7feea34c6c6d in Base::InterpreterSingleton::runCommandLine(char const*) f                                                                                                                       rom /usr/lib/freecad-daily-python3/lib/libFreeCADBase.so+0x1d
#61  freecadcmd-daily(+0x551e) [0x55ddd180e51e]
#62  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7feea298cd90]
#63  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7feea298ce40]
#64  freecadcmd-daily(+0x5ac5) [0x55ddd180eac5]

luzpaz, 2024-09-10:
Can you provide the output for `FreeCADcmd --version --verbose`

c3kkos, 2024-09-10:
Of course:
```

➜  freecad_macros freecadcmd-daily --version --verbose
FreeCAD 1.0.0 Revision: 38684 (Git)

Library versions:
boost    1_74
Coin3D   4.0.0
Eigen3   3.4.0
OCC      7.8.1
Qt       5.15.3
Python   3.10.12
PySide   5.15.2
shiboken 5.15.2
SMESH    7.7.1.0
VTK      9.1.0
xerces-c 3.2.3

```

luzpaz, 2024-09-10:
> As it's not possible to create or access the [view provider](https://wiki.freecad.org/Viewprovider) in headless mode. What's possible is to load FreeCADGui in headless mode but there is no way to access the GUI document because it won't be created and consequently there exist no view providers.

>However, what's possible is to create a [scenegraph](https://wiki.freecad.org/Scenegraph) representation of an object

Per the documentation in https://wiki.freecad.org/Headless_FreeCAD#Scenegraph_Representation 

However, FC should fail gracefuly instead of crash so this seems like a valid bug

c3kkos, 2024-09-10:
Yes, from the python interpreter, no output is shown, but this can be not a concern for FC.

I'm afraid i need to apologize for not being able to focus on the point:

i don't need and do not want to mess with the Gui, using CLI.
I thought using PartDesignGui to be a valid workaround but i failed.

I need to use PartDesign and i can't!

```
➜  freecad_macros python3 test.py
Traceback (most recent call last):
  File "/home/c3kkos/freecad_macros/test.py", line 8, in <module>
    import PartDesign
ModuleNotFoundError: No module named 'PartDesign'
```

luzpaz, 2024-09-10:
Can you provide the external script that crashes?

c3kkos, 2024-09-10:
Yes! here it is!
```

FREECADPATH= '/usr/lib/freecad-daily-python3/lib'
import sys
sys.path.append(FREECADPATH)

import FreeCAD as App
import Part
import PartDesign

doc = App.newDocument()

App.ActiveDocument.addObject('PartDesign::Body','Body')

App.ActiveDocument.addObject('PartDesign::AdditiveBox','Box')
App.ActiveDocument.getObject('Box').addObject(App.ActiveDocument.getObject('Box'))

App.ActiveDocument.recompute()

App.ActiveDocument.saveAs('test.FCStd')

```

Syres916, 2024-09-10:
With regard to the original report, replace the top three lines with:

import sys

```
FREECAD_PATH = 'usr/lib/freecad-python3/lib' #needed for C++ modules or FreeCAD (e.g. 'FreeCAD.ActiveDocument')
FREECAD_MOD_PATH = 'usr/lib/freecad/Mod' #needed for PartDesign and other Python modules

if FREECAD_PATH in sys.path:
    pass
else:
    sys.path.append(FREECAD_PATH )

if FREECAD_MOD_PATH in sys.path:
    pass
else:
    sys.path.append(FREECAD_MOD_PATH )

```

c3kkos, 2024-09-10:
Many thanks.

That should be added in wiki! 

i googled a lot, and never found a MOD PATH include. (my fault anyway).

glad at least we found an unexpected crash here.

mwganson, 2024-10-03:
You don't need the gui to use Part Design in a headless script.

This worked for me in a test:

```
doc = FreeCAD.newDocument()
doc.FileName = "test.FCStd"
body = doc.addObject("PartDesign::Body","Body")
box = doc.addObject("PartDesign::AdditiveBox","Box")
body.addObject(box)
doc.save()
```
Method (on Windows)

1) Open the bin folder in file explorer
2) Ctrl+Shift+RMB click -> open power shell here
3) cmd <enter>
4) I put the test macro file in the bin folder for convenience to not have enter the full path in the command line
5) FreeCADCmd name_of_macro.FCMacro

After it exited I found the test.FCStd file in the bin folder.  Opening it I found the body and the pad inside.  They were both invisibile, but they were there.

closed 2025-06-04 by chennes