FreeCAD/FreeCAD #16408: BIM: IFC: wrong mouse cursor in the 'IFC import options' dialogue

semhustej, 2024-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Priority: low - . The problem does not have influence on the functionality of the dialogue.

When opening IFC file in 1.0RC on Windows, I get wrong mouse cursor type (send timer instead of arrow)
[image]

Steps to reproduce:

1. Open IFC file
2. Wrong cursor is displayed

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-10:
The cursor is set by FreeCAD when starting an open/import command. Indeed while on the options dialog we could change this to the default arrow one.

semhustej, 2025-07-21:
The issue is still valid.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42605 (Git)
Build date: 2025/07/18 16:20:04
Build type: Release
Branch: main
Hash: da5e18f5d8e7eb4df89c86c9562599435b5c5187
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * OldReporting
  * Reporting
  * Telemetry 1.0.1

```

semhustej, 2025-11-21:
The issue is still valid.