FreeCAD/FreeCAD #16409: BIM: Error when Arch_Wall is created on top of Group

semhustej, 2024-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When Group is selected (or IfcProject container) and Arch_Wall command is started, FreeCAD outputs an error. See behavior in the video - the behavior is dependent on the content of the group..

Similar result happens when testing with BIM_slab tool and I suspect other BIM elements.

Steps to reproduce:

1. Select a group or IfcProject conatainer
2. Start Arch_Wall command
3. Error shows up

https://github.com/user-attachments/assets/401a0cba-271e-45c6-beaf-391dcadad4bd

Error code:
`pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_38643\Mod\BIM\ArchWall.py", line 266, in execute
    extdata = self.getExtrusionData(obj)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_38643\Mod\BIM\ArchWall.py", line 721, in getExtrusionData
    normal = DraftGeomUtils.get_shape_normal(obj.Base.Shape)`

Test file:
[IfcProject-test_001.zip](https://github.com/user-attachments/files/16944597/IfcProject-test_001.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

BIM

### Anything else?

When trying to produce the same error with Draft Layer or 'Layers' group, I get the correct behavior: nothing happens and FC outputs
`Walls can only be based on Part or Mesh objects`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-09-10:
A group is not a "physical" object. It is only a virtual way of cataloging.
Therefore, you cannot build a wall based on a group. You need to select lines inside the group.
I think this is not a mistake, but I think there should be a convenient way to select objects inside a group.

I really miss a method like in blender (selecting all objects inside a collection. This is an analogue of a group)
For some reason, selecting all instances in freecad does not work. You have to make a selection using multiple clicks with ctrl or shift

https://github.com/user-attachments/assets/28d7cb58-44c1-4124-8207-704fa98b327b

semhustej, 2024-09-10:
Hi @kaiwas,

I understand that when I select the group, it is not selecting elements inside the group. The issue was created because I think the software should not proceed with the Arch_Wall command, when a group is selected - especially when it leads to errors and corruption of the file. The appropriate behavior would be to report 'The command can not be based on group object' - the same way it does with layers.

semhustej, 2024-09-10:
@kaiwas 
the tool you need to use to select all instances in the group is Draft_SelectGroup:
https://wiki.freecad.org/Draft_SelectGroup
It might be useful to have this under the right-click menu of the group, you might suggest a new feature :)

the Std_SelectAllInstances that you are referring to selects all instances of the same element in the project tree:
https://wiki.freecad.org/Std_TreeSelectAllInstances

kaiwas, 2024-09-10:
> the tool you need to use to select all instances in the group is Draft_SelectGroup:
> https://wiki.freecad.org/Draft_SelectGroup

It's hard to believe, but until you mentioned this tool, I hadn't noticed it.
For some unknown reason, all the selection options are scattered around in different places.
I think this needs to be structured somehow.
Thanks!

semhustej, 2024-09-10:
> It's hard to believe, but until you mentioned this tool, I hadn't noticed it. For some unknown reason, all the selection options are scattered around in different places. I think this needs to be [...]

Happy to promote the tool. I discovered it only because I was looking into Draft Layers and the tool button is on the same panel. I just found out about 'select all instances' when you mentioned it :)

yorikvanhavre, 2024-09-11:
Indeed BIM creation tools should not proceed is the selected object is a group. I think we can call that a bug. I'll look into it

paullee0, 2024-10-14:
> @kaiwas the tool you need to use to select all instances in the group is Draft_SelectGroup: https://wiki.freecad.org/Draft_SelectGroup It might be useful to have this under the right-click menu of the group, you might suggest a new feature :)
 
Just dropped into this discussion, probably off-topic and may need a separate issue.  Before that, I have a look at the wiki, it indicated BIM WB but seems not found in FreeCAD (yet?).  Then, should a shortcut better?  Feel free to discuss separately.

Back to original topic, agreed with Yorik that this should be considered as a bug :)

semhustej, 2024-10-14:
> Just dropped into this discussion, probably off-topic and may need a separate issue. Before that, I have a look at the wiki, it indicated BIM WB but seems not found in FreeCAD (yet?). Then, should a [...]

The Draft_SelectGroup is in fact under the the RMB context menu in 1.0RC, but only in the BIM workbench. When you switch to different workbench, it is not there. I plan to file an issue asking to add this section of RMB menu to Draft workbench, because this seems very useful and intuitive.
In draft WB, the tool is on the Draft utility toolbar too.
In BIM WB the tool is also under the Utils menu, as described on wiki.

BIM WB

[image]

paullee0, 2024-10-14:
> The Draft_SelectGroup is in fact under the the RMB context menu in 1.0RC, but only in the BIM workbench. When you switch to different workbench, it is not there. I plan to file an issue asking to [...]

Looks good to me.  Better still for me if a shortcut could be developed.

> In BIM WB the tool is also under the Utils menu, as described on wiki.
Found, so many items there, thanks :)

yorikvanhavre, 2024-10-15:
The utils menu is a mess indeed, we'll need a discussion to organize that better later on :sweat_smile:

PR #17264 (2024-12-02): BIM: ensure the Base has the correct type
Fixes #16409
commit 8ba34888 (2024-12-02): BIM: ensure the Base has the correct type (#17264)

* BIM: ensure the Base has the correct type - fixes #16409

* BIM: refactor according to chenne's comment
closed 2024-12-02 by chennes