FreeCAD/FreeCAD #16411: Sketcher: Enabling grid in sketcher view results in crash

Sadia2000, 2024-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When in sketcher workbench and editing a sketch, toggling the grid via the UI results in a crash.

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-10:
Can you install the dev version (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) and check if this is reproducible? (We are on the verge of releasing 1.0 so there are no backport fixes happening for 0.21.x)

xtemp09, 2024-09-10:
Toggling the grid does not cause crash on my platform.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38689 (Git)
Build type: Debug
Branch: main
Hash: 3503d677a91c98672b80c2b312593440fb3da0db
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
```

Sadia2000, 2024-09-10:
Oh, I would try it out and let you know soon. Thanks for the quick response :D

luzpaz, 2024-09-10:
Closing. If reproducible on dev, then please re-open. TIA

closed 2024-09-10 by luzpaz