FreeCAD/FreeCAD #16419: Rotations of piece not exporting to STL

chuckv3, 2024-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I could swear that in previous versions... if I rotated a piece and then exported it that the rotation would appear in the export. Now when I rotate a piece and export it, the STL file (when imported into MatterControl anyway) looks as if I had not rotated it.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-10:
Hi @chuckv3 can you test on the latest 1.0RC to see if you can reproduce? 
https://blog.freecad.org/2024/09/10/the-first-release-candidate-of-freecad-1-0-is-out/

luzpaz, 2024-09-11:
@chuckv3 please respond

Roy-043, 2024-09-13:
This is a known issue in V0.21.2 that has already been fixed.

closed 2024-09-13 by Roy-043