FreeCAD/FreeCAD #16422: [V1.0 RC1/Sketcher] No visual feedback when external line is selected

DerMouse, 2024-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the sketcher workbench, when using external reference lines (not only straight lines, anything), the lines are properly highlighted in the preselection mode color, but do not change color when selected. External points are properly changing color when selected.

### Full version info

```shell
OS: Gentoo Linux
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * freecad.gears 1.2.0
  * persistenttoolbars
```

### Subproject(s) affected?

Sketcher

### Anything else?

Preselecting the line (configured preselection color is light blue here)
[image]

Selected a point from the external geometry
[image]

Selected the same vertical line as in the first image (believe me : the line IS selected)
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-11:
I've narrowed the regression down to 224 commits and a Git Bisect is underway, looks like sometime late July/early August time when the selection mechanism for external geometry stopped working correctly.

kadet1090, 2024-09-11:
https://github.com/FreeCAD/FreeCAD/pull/16355 this PR should fix it probably?

Syres916, 2024-09-11:
For completeness, I couldn't compile all of the commits so the best I can say is [PR15509](https://github.com/FreeCAD/FreeCAD/pull/15509) was the trigger of the regression and while https://github.com/FreeCAD/FreeCAD/pull/16355 fixes the external geometry selection, I've found that a new set of [errors](https://github.com/FreeCAD/FreeCAD/pull/16355#issuecomment-2344246162) get raised on some 'real' geometry.

CalligaroV, 2024-09-11:
Thanks @Syres916 for trying #16355 and @kadet1090 for linking it to this issue!

@Syres916, as you may have already noticed I'm participating to the discussion about the errors you found in the conversation of #16355.
Regardless for that, and considering your feedback, I'll add that PR as a fix for this issue.

Thanks again!

closed 2024-09-13 by chennes

DerMouse, 2024-09-14:
With latest Freecad weekly build :

OS: Gentoo Linux
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.2.0
  * persistenttoolbars

this bug is solved. Thanks !

PR #16355 (2024-09-13): Sketcher/Toponaming: import SketchObject::checkSubName() from LS3
related to https://github.com/FreeCAD/FreeCAD/issues/13177#issuecomment-2307792664

the imported code restores the synchronization of the selection of external edges between the 3D view and the Task Panel

as for https://github.com/FreeCAD/FreeCAD/issues/16422#issuecomment-2344255862, this PR fixes #16422