FreeCAD/FreeCAD #16425: Bad readability in the Tree View

etmd, 2024-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With default parameters, the tree view texts are not easy to read. The texts are white with a dark border, on a light background, which makes it hard to read them.
Besides, when clicking the "eye" to make the background opaque, the background turns white, and the texts are not readable anymore.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/Switzerland (fr_CH)
Installed mods: 
  * A2plus 0.4.61
  * Curves 0.6.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-10:
You think this is a QtStyle issue, @MisterMakerNL ?

MisterMakerNL, 2024-09-11:
This is usdye where the correct overlay stylesheet isn't applied. Or a bug where the stylesheet needs a programming reboot.
But since the colours are changes from default I guess wrong stylesheet? Overlay light background/light theme is probably correct here.
And yes, this could be made a lot more user friendly.

etmd, 2024-09-11:
Seeing your answers, this didn't seem normal... So I tried to revert all settings to default, chose the dark theme, then switched back to light theme, and the text is now black.

So it seems that the theme did not properly "refresh". Maybe an interference between the official FreeCAD release and the RC1 ?

Anyway, I may have posted this issue for nothing, since the software does not seem to have a problem. Sorry for the inconvenience...

Syres916, 2024-09-11:
The root cause is that the FreeCAD Classic theme is selected for new users in the First run Start Widget which means the code to apply the FreeCAD Classic.cfg and post.FCMacro isn't activated and therefore the Overlay is left at Auto instead of Light Theme + Dark Background.qss. If the user clicks any of the other two themes and then goes back to FreeCAD Classic the settings are applied.

Syres916, 2024-09-11:
PR https://github.com/FreeCAD/FreeCAD/pull/16439 submitted

gjpc, 2024-10-13:
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * AirPlaneDesign 0.4.0
  * BOLTSFC
  * CurvedShapes 1.0.9
  * Curves 0.6.37
  * fasteners 0.5.21
  * freecad.gears 1.0.0
  * Help 1.0.3
  * parts_library
  * Plot 2022.4.17
  * slic3r-tools

```
Switching modes never changes the color of object tree text. It is always white. With this version only dark mode is usable.

maxwxyz, 2024-10-29:
@hyarion is this also solved by your PR?

closed 2025-08-05 by github-actions[bot]