FreeCAD/FreeCAD #16426: Apparent performance regression.

GAZ082, 2024-09-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have RC1 and another build some kind folk at the Discord shared. I'm doing a View Fit on both. You see that the animation is not smooth, there are "pulls". Also if you check the top right corner of my video and see the CPU usage, it's like 10% higher than the old build.
Settings used are the same. Both are APPIMAGES.

https://github.com/user-attachments/assets/6514ef28-f919-464c-ba47-d4b7fe29e667

### Full version info

```shell
## RC1:
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Render 2024.8.4
  * OpenTheme 2024.9.1
  * QuickMeasure 2022.10.28

## 0.22 compared:
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git) AppImage
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Render 2024.8.4
  * OpenTheme 2024.9.1
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GAZ082, 2024-09-10:
I did a VF here, but any kind of model manipulation is choppy. @luzpaz

luzpaz, 2024-09-10:
Added RC-Blocker tag to make sure this is highlighted.

GAZ082, 2024-09-14:
Still with issues in 1.1.0dev
Wonder if this is AMD card specific. I have a  **AMD Radeon™ RX 550 / 550 Series**
@skrystosik  has a Mac with xr6600xt
@luzpaz

maxwxyz, 2024-09-14:
@luzpaz I've changed your tag RC-blocker to blocker as RC is already out.
@chennes FYI

Rexbas, 2024-09-15:
I have a NVIDIA RTX 3060 Ti, running driver 560.35.03 with Wayland and it works fine.

skrystosik, 2024-09-15:
> I have a NVIDIA RTX 3060 Ti, running driver 560.35.03 with Wayland and it works fine.

probably banku you have windows. Problem is on OSX

leoheck, 2024-09-15:
I think I observed poor performance on RC1 too. Not as hard as in the video above but something was not as fast as before. It tough ot was my design that was getting bit big and complex. 

I have NVIDIA but I am using the Nouveau drivers. I am running X11. 

I observed a worse behavior when moving the 3d model when working with Trackpad mode, using Draft Polyline.

GAZ082, 2024-09-15:
This is ridiculous. I renamed my $HOME/.config/FreeCAD files to .old. Restarted, the performance was ok. I renamed back the SAME FILES to restore my configuration, expecting to see again the choppyness, and IT KEPT WORKING FINE.

I do not want to make any conjecture, but this is the second time that messing with the config files "resolved" the issue.

@leoheck see if this works for you as well.

GAZ082, 2024-09-15:
@skrystosik see if my "solution" worked for you too. Rename the config files to .old, restart FC, see if works fine. And now revert back the .old to the original name to get back  your config options and see if the good performance sticks after restarting a second time.

leoheck, 2024-09-15:
I see you have this `OpenTheme 2024.9.1`, I am also using open theme, maybe it is related...

skrystosik, 2024-09-16:
> @skrystosik see if my "solution" worked for you too. Rename the config files to .old, restart FC, see if works fine. And now revert back the .old to the original name to get back your config options and see if the good performance sticks after restarting a second time.

Actually after going back to 0.22 and upgrade to 1.0 RC that was out 3 days ago I don't see this issue anymore. I have only choppines while using space mouse without dominant mode but this is probably because of zoom step. Without dominant mode it's constantly changing even if I don't zoom and this makes this not smooth. I'll use to use dominant mode and I'm happy for now.

chennes, 2024-09-16:
I'm dropping the Blocker tag here since it seems to be a strangely intermittent (and therefore difficult to reproduce and debug). Issue. I will leave the Issue open, but the devs at this morning's meeting didn't feel like this should stop the release of 1.0.

leoheck, 2024-09-16:
I don't have this choppy animation. But I am experiencing a reduced speed while designing. It was not like that 2 or 3 weeklies before.

GAZ082, 2024-09-21:
Yeah, i'm also noticing this.

luzpaz, 2024-09-25:
JFYI ,speed optimization in Assembly wb pending #16787  
Not convinced it's totally related but it could be ?

luzpaz, 2024-10-11:
Can we get more eyes on this to fish out the intermittency of this bug ?

ChrisJones79, 2024-10-19:
Don't know if this helps but I get a smooth animation with Nvidia latest propietary drivers on debian+KDE. Buuuut Fit All smoothly animates to final orientation and Fit Selection jumps to final orientation. Also, there appears to be no difference in final orientation. Meaning any face on the object using Fit Selection produces the same final image as just using Fit All.
Here's my build info:
```
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

```

luzpaz, 2024-10-23:
@ChrisJones79 can you also check the weekly build, if you don't mind? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) TIA

ChrisJones79, 2024-10-23:
1.0.0RC2.38806

Is that what you're looking for?

luzpaz, 2024-10-23:
@ChrisJones79 that is Release Candiddate. I'm asking if you would kindly test the development version (which we call 'weekly' but is actually generated more that once a week)

ChrisJones79, 2024-10-23:
Ok, I understand. Yes, I'll do that. Is there a way to automatically update, or, at least be notified when a new build is ready?

GAZ082, 2024-10-23:
@ChrisJones79 i see you do. You may want to change or delete the "notify send" commands, dont know if these work for KDE (gnome user here).

```#!/bin/bash

clear
cd /tmp
binaryLocation="/opt/FreeCAD"
binary="/opt/FreeCAD/FreeCAD.AppImage"
remoteFileURL=$(curl https://api.github.com/repos/FreeCAD/FreeCAD-Bundle/releases/tags/weekly-builds | jq '.assets[].browser_download_url' | grep -Eo 'https.*Linux-x86_64.*AppImage' | head -n1)
remoteFile=$(echo $remoteFileURL | grep -Eo "FreeCAD_.*AppImage$")
# remoteFileURL="https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-37100-conda-Linux-x86_64-py311.AppImage"
remoteVer=$(echo $remoteFileURL | grep -Eo "[0-9]{5,}")
touch $binaryLocation/last_version_downloaded
localVer=$(cat $binaryLocation/last_version_downloaded)
if [ "$remoteVer" = "$localVer" ]; then
    echo "-------------------------"
    echo "-Running latest version!-"
    echo "-------------------------"
    $binary
    else
    notify-send "Updating FreeCAD"
    echo "-------------"
    echo "-Updating...-"
    echo "-------------"
    echo $remoteFileURL
    wget $remoteFileURL
    echo $remoteVer > $binaryLocation/last_version_downloaded
    mv $remoteFile $binary
    chmod +x $binary
    notify-send "Done."
    $binary
fi```

ChrisJones79, 2024-10-23:
Thank you! I'll check out the script this evening.

GAZ082, 2024-11-01:
This issue emerged again. There is something in the config that triggers this. Used the Safe Mode and works fine. (Black mode my customized one, white is Safe Mode).

OS: Arch Linux (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

https://github.com/user-attachments/assets/0098b880-1cf3-4aac-83b7-294329da9c78

luzpaz, 2024-11-01:
Can you attach the backed-up config file in question ?

GAZ082, 2024-11-02:
I suppose you refer to my actual config files. Yes. 
[files.zip](https://github.com/user-attachments/files/17606804/files.zip)

I'm trying to figure out a way  to compare with the vanilla files Safe Mode is using.

@luzpaz

ChrisJones79, 2024-11-09:
@luzpaz do you happen to run yours as a scheduled job? cron? Or do you manually run this?

Also, what is the dev release frequency?

> @ChrisJones79 i see you do. You may want to change or delete the "notify send" commands, dont know if these work for KDE (gnome user here). And update `binaryLocation`.
> 
> ```
> 
> clear
> cd /tmp
> binaryLocation="/opt/FreeCAD"
> binary="/opt/FreeCAD/FreeCAD.AppImage"
> remoteFileURL=$(curl https://api.github.com/repos/FreeCAD/FreeCAD-Bundle/releases/tags/weekly-builds | jq '.assets[].browser_download_url' | grep -Eo 'https.*Linux-x86_64.*AppImage' | head -n1)
> remoteFile=$(echo $remoteFileURL | grep -Eo "FreeCAD_.*AppImage$")
> # remoteFileURL="https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-37100-conda-Linux-x86_64-py311.AppImage"
> remoteVer=$(echo $remoteFileURL | grep -Eo "[0-9]{5,}")
> touch $binaryLocation/last_version_downloaded
> localVer=$(cat $binaryLocation/last_version_downloaded)
> if [ "$remoteVer" = "$localVer" ]; then
>     echo "-------------------------"
>     echo "-Running latest version!-"
>     echo "-------------------------"
>     $binary
>     else
>     notify-send "Updating FreeCAD"
>     echo "-------------"
>     echo "-Updating...-"
>     echo "-------------"
>     echo $remoteFileURL
>     wget $remoteFileURL
>     echo $remoteVer > $binaryLocation/last_version_downloaded
>     mv $remoteFile $binary
>     chmod +x $binary
>     notify-send "Done."
>     $binary
> fi```
> ```

JohnOCFII, 2024-11-09:
> Also, what is the dev release frequency?

For the past many months, the "weekly" development releases have happened twice a week.  Usually late Monday and late Friday (US Central time -- UTC-6).  For example, I see fresh releases about 13 hours ago. https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

ChrisJones79, 2024-11-10:
Thank you @JohnOCFII - I had other questions but it's not really in the scope of this issue so I'm good for now. 

One caveat for the script, for anyone reading later, is to have a dev version already installed. I changed my OS and the script needs an binary at the $binarylocation.

> For the past many months, the "weekly" development releases have happened twice a week. Usually late Monday and late Friday (US Central time -- UTC-6). For example, I see fresh releases about 13 [...]

GAZ082, 2025-04-12:
I tried to replicate this in 1.0 and in 1.1 in both Safe and standard mode and was not possible, it is snappy.

@maxwxyz i think we can close this one.

closed 2025-04-12 by luzpaz