FreeCAD/FreeCAD #16431: UI/UX: Importing SVG window

leoheck, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

First of all, the title of the pop up window is a bit off.
Suggestion: `SVG Import Settings`

[image]

Second, it seems that the `Open SVG as` functions are switched.
Selecting the first one `Image formats (Freecad)`, imports an image, but it is written (Freecad).
The second option `Import geometry (importSVG)`, create some Part Objects.

This text is a bit misleading, and should be improved.
I have no suggestion since I am not sure if the functions of each itens are right. 

Extra: Tool-tips would be good to have too.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-11:
The dialog informs that there are 2 modules (FreeCAD and importSVG) that can import an SVG file. The same dialog is used whenever more than one import module is available for a file format.

leoheck, 2024-09-11:
What are those modules? Code? I don't think those names make the end user to understand the operation of each item better.

luzpaz, 2024-09-11:
How do we give the end-user here a context to each choice ? Should we have links to the docs here ?

leoheck, 2024-09-11:
I think the options should be rewritten to be clear, removing those module names. It the module names are important, then adding something as a tool-tip that explains that would be enough.

I would suggest something....

Open SVG as...
- `an Image`
- `a Part object`, `a Freecad object`, `a Freecad geometry`

luzpaz, 2024-09-12:
@Roy-043 what say you ?

Roy-043, 2024-09-12:
Core is displaying the information it has available. Which basically matches what you see in the Std_Import dialog. For me this is enough. It is also consistent I would say.

Module names should stay in the dialog IMO. You might have two modules that can import DXF files, where their results are similar, but in some cases one works better than the other.

ElementW, 2025-08-13:
@leoheck This was autoclosed but the situation is unchanged as of `1.1.0dev.42885` (2025/08/08 15:11:51), could you reopen?

@Roy-043 If this is the result of Core displaying what information it has, then it means that information is insufficient for a user (who doesn't know FC's inner machinations) to make an informed decision. Keeping the module names in that window is not incompatible with having more descriptive labels. One could even extend this with icons to illustrate what is going to happen, for example:

[image]
[image]

leoheck, 2025-08-13:
@luzpaz could you reopen this issue for us?

leoheck, 2025-08-13:
@ElementW can you use a square with round corners. and a darker image bg?

ElementW, 2025-08-13:
@leoheck Like this?

[image]

[image]

ElementW, 2025-08-13:
I only focused on adding example icons but hadn't given more thought the whole dialog window itself.

This would already be a lot better as you're immediately presented with a representation of each option, there's more space to add text if needs be (or even a config sub-button), it's much more forgiving to cursor position/better on tablet input, is easier to translate as strings aren't broken up/context sensitive, and saves a click.

[image]