FreeCAD/FreeCAD #16432: BIM: Missing element name prefix id

BearCarpenter, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create Empty File, Go to Part Design Workbench, add Body, add Sketch, add rectangle, use Pad and save it in part_library location, Close file
2. Create Empty File, Go to BIM Workbench, import created body by Object library

Raport view:
<ElementMap> BimLibrary.py(581): Missing element name prefix id

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-11:
Confirmed. But it is not a BIM issue. It is a `mergeProject` issue that also occurs when you use the Std_MergeProjects command.

luzpaz, 2024-09-11:
cc @bgbsww :point_up_2:

bgbsww, 2024-09-11:
Interesting.  So this is because we are trying to restore the element map of one file into another, and that isn't valid.

That will likely require an ignoring of the element map in the file being merged, and then a recompute to generate new entries.

Does this happen in LinkStage3?  That will sort out whether this is a code porting problem or a code writing problem.

BearCarpenter, 2024-10-31:
@PaddleStroke @yorikvanhavre
You may have corrected this in the last release.
Here is similar problem [16436](https://github.com/FreeCAD/FreeCAD/issues/16436)
Both works! Am i right?

maxwxyz, 2024-11-09:
Closing then.

closed 2024-11-09 by maxwxyz

maxwxyz, 2024-12-27:
Acc. to the forum report https://forum.freecad.org/viewtopic.php?p=799848#p799848 this seems to be an issue in 1.0

ErtugrulEge, 2025-02-10:
Now when I insert an object from 0bjects library I have this message 

` <ElementMap> BimLibrary.py(586): Invalid element name string id`

[image]

FEA-eng, 2026-09-26:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```