FreeCAD/FreeCAD #16435: DXF importer: AttributeError when importing dimensions

furgo16, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Start FreeCAD
2. Create a new document
3. File > Import...
4. Select the DXF file attached
5. Use the import settings in the screenshoot below. Most importantly, check `Use legacy Python exporter` (as the C++ importer does not yet support importing dimensions) and `Texts and dimensions`.
6. Click OK
7. The DXF file entities are imported, but dimensions are not. The Report view shows the following error:

```python stacktrace
opening $HOME/Documents/Layertest.dxf...
drawing 8 lines...
drawing 1 polylines...
drawing 2 circles...
drawing 2 dimensions...
pyException: Traceback (most recent call last):
  File "<string>", line 5, in <module>
  File "/snap/freecad/1131/usr/Mod/Draft/importDXF.py", line 2848, in insert
    processdxf(doc, filename)
  File "/snap/freecad/1131/usr/Mod/Draft/importDXF.py", line 2599, in processdxf
    lay.addObject(newob)
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'addObject'
```

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git) Snap 1131
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Installed mods: 
  * OpenTheme 2024.9.1
  * Curves 0.6.46
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-11:
Confirmed. The file contains a local `addObject` function that has not been implemented everywhere for some reason.

yorikvanhavre, 2024-09-12:
> sometimes addObject is called as a local function defined in this module, sometimes as a method from a parent class.

Some objects, those derived from App::DocumentObjectGroup or those that have the App::GroupExtensionPython extension, have an addObject method built-in. Others don't, so for those we often write our own

furgo16, 2024-09-12:
Would it make sense to a) modify the local `addObject` function to handle objects with a built-in `addObject` method and b) to always call the local the local `addObject` function within the module?

In a way, this is already done if the function is called with a layer:

https://github.com/FreeCAD/FreeCAD/blob/614a14fe87af6ea0a978ee6312b298e4033f1839/src/Mod/Draft/importDXF.py#L1921-L1935

yorikvanhavre, 2024-09-12:
> Would it make sense to a) modify the local addObject function to handle objects with a built-in addObject method and b) then always call the local addObject function within the module?

Yes I think that's a good idea

furgo16, 2024-09-12:
Ok, I'll try to give that a go on a separate PR, after https://github.com/FreeCAD/FreeCAD/pull/16469 is reviewed/merged. 

There are other entities in the module that might need the same fix, and it might be a good opportunity to explore if there is room for optimization.

PR #16469 (2024-09-12): DXF importer (Python): allow importing dimensions
Fixes #16435. With this change, the legacy DXF importer can now import dimensions.

Notes:
1. <strike>As far as I can tell, the layer class does not have an `addObject()` method, or at least I did not find any documentation. I added the check nevertheless, as it was done in other places of the module. It would be great to have confirmation from a maintainer if it's needed.</strike> Update: I think I can answer myself the question, thanks to a comment in the code: `For old style layers, which are just groups `. So the check for `addObject()` is  [... cut]
commit 5fa7fb75 (2024-09-12): Check location and call corresponding addObject, add missing import. Fixes #16435
closed 2024-09-12 by yorikvanhavre