FreeCAD/FreeCAD #16436: Assembly: Unnamed#[joint type] is unconnected to a grounded part so it is ignored.

BearCarpenter, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

> Each command was triggered with the mouse, but I am writing specific commands here for easier understanding

1. First:
- Create `Empty File`, 
- add `Assembly_CreateAssembly`,
- add `Cube` and `Cylinder` from `Part Workbench`,
- select `Cube` and use `Assembly_ToggleGrounded`,
- add `Assembly_CreateJointDistance` between flat faces of `Cube` and `Cylinder`,
- Save file and Close
2. Next:
- Create `Empty File`,
- add `Assembly_CreateAssembly` (`base assembly`),
- add `Cube` and use `Assembly_ToggleGrounded` ,
- add previous created file/assembly by Std_MergeProjects (`File`->`Merge domcument...`),
- move by mouse `imported Assembly` into `base assembly` in `Model Tree View`,
- add `Assembly_CreateJointDistance` between faces of
  - grounded `Cube` 
  - and `Cube from imported Assembly`.

Raport view:
`Unnamed#Distance is unconnected to a grounded part so it is ignored.`

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-10-08:
This will be fixed by https://github.com/FreeCAD/FreeCAD/pull/15629

BearCarpenter, 2024-10-11:
Currently for the same operations freecad crashed for this version:
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git) AppImage
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

closed 2024-10-28 by yorikvanhavre