FreeCAD/FreeCAD #16437: BIM: 1.0RC1: lines/polylines are not imported from during IFC import

semhustej, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening IFC file with lines, unfilled polylines and unfilled rectangles, these elements are not imported properly in FC 1.0RC1. It seems like a regression, because when testing with FC 0.21.2 all lines are imported properly.

Sample file (originally exported from FC):
[Test_ Draft-Lines_IFC_01-Level-paullee.zip](https://github.com/user-attachments/files/16961592/Test_.Draft-Lines_IFC_01-Level-paullee.zip)

How the file opens in BonsaiBIM:
[image]

How the file opens in FreeCAD (tested with different import options):
[image]

**Probably related issue:**
When working in strict IFC mode, user is unable to create lines, unfilled polylines, rectangles or circles. Filled rectantgles and circles can be created. See behavior in video:
https://github.com/user-attachments/assets/a4c6499f-7557-492b-90aa-421e21531a20

Related forum discussion: https://forum.freecad.org/viewtopic.php?t=90474

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-13:
Indeed the new (nativeIFC) IFC importer does not support 2D objects just yet. Just a bit of patience and we'll be there :)

In the meantime, you can still use the old importer, as explained in https://wiki.freecad.org/Arch_IFC#The_Arch_importer

luzpaz, 2024-09-13:
@yorikvanhavre is this a dupe ?

yorikvanhavre, 2024-09-13:
> @yorikvanhavre is this a dupe ?

Not really... It's more a "known bug" or "known-that-it-is-not-yet-implemented" :) Basically it's one of the items of my NativeIFC roadmap at https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/doc/roadmap2024.md but that's not really valid as a duplicate...

semhustej, 2024-09-13:
@yorikvanhavre  Thank you for the clarification. If the feature is on the roadmap, this issue can be closed I guess.

yorikvanhavre, 2024-09-16:
Let's keep it open because it's still a valid bug, and it'll be satisfactory to close it when it's done :sweat_smile:

semhustej, 2025-07-21:
The feature has been added, I am able to import the 2D geometry. Thank you.
In my opinion this issue can be closed.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42531 (Git)
Build date: 2025/07/11 16:00:50
Build type: Release
Branch: main
Hash: 759bad7455a3a554cdf48a8182709cad991c8fe5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * OldReporting
  * Reporting
  * Telemetry 1.0.1
```

closed 2025-08-11 by semhustej