FreeCAD/FreeCAD #16441: CAM workbench adding PropertyBag to custom tool

oodanner, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the CAM workbench if you select the body of a custom tool you have created and then add a propertybag to it with the menu items: Cam, utils, propertybag, it creates an attributes entry in the Model.  If you single left click on the attributes item then you get a data box below and if you right click in there, you can add a property to the attributes(attached ProblemArea.PNG).  Somehow this box isn't tied to the actual attributes and if you try to save it and add this tool it will tell you that you don't have a PropertyBag in the console when you try to load it as a tool.  I found that you have to actually double click the attribute item which takes you to a another page where you can add properties(attached DoubleClickAttributes).  If you create an attribute the first way I mentioned then double clicking on the attribute will show up no attributes even though you have created one.  Hopefully this is an easy fix.
[image]
[image]

Thank you all for the awesome work you have been doing!  With RC1 this is now my go to modeling application with your sketcher improvements!

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-11:
Do you happen to know if this is a regression bug ?

oodanner, 2024-09-11:
If I understand correctly what a regression bug is then I think you are
asking if this is a new problem in this release and I have tested other
versions before this and it has been there as well so I think this has been
a problem for a while.

spanner888, 2024-09-11:
@oodanner  You have described adding a Property Bag to a Custom Tool that you have already created and you mentioned doing this in the CAM workbench.

The correct way to create a custom ToolBit Shape is done using the PartDesign workbench, including adding the PropertyBag.

If you meant creating a new ToolBit from an existing ToolShape, then the ToolShape would already contain the PropertyBag.

Could you please attach the ToolShape and any ToolBit you are creating from the ToolShape, and ideally clarify the exact process you are attempting. 

The wiki [CAM ToolBitLibraryOpen](https://wiki.freecad.org/CAM_ToolBitLibraryOpenl) might help, see Manage Toolbits section.  Also usefull is the section ToolBit architecture in [CAM Workbench](https://wiki.freecad.org/CAM_Workbench)

oodanner, 2024-09-12:
I guess I didn't add enough detail to the bug or explain it clearly
enough.  I know how to create the ToolBit shape and I'm not aware of any
way to add the PropertyBag from the PartDesign workbench as there is no
option to do that from the menu bar.  You have to be in the CAM workbench
to even get the menu option for adding a PropertyBag.  Did the files I
attached not come through on the bug?  I thought it was pretty clear that I
can and do know how to create the PropertyBag and have successfully done so
with the PartDesign and then by adding the PropertyBag to the body from the
CAM workbench.  The problem is that if you don't double click the
attributes in the model list and you instead right click the data area
after just highlighting the attributes then whatever you create in there
doesn't actually save to the PropertyBag.  However if you double click the
attributes and add it from that screen then it does save correctly to the
PropertyBag and it all works as expected.  I had hoped the screenshots I
provided would be enough to explain the bug.

spanner888, 2024-09-12:
Also my bad explanation and sorry as I thought you had somehow created the toolbit in CAM WB.

I can see two images, but no files.

Have you seen this page [CAM ToolShape](https://wiki.freecad.org/CAM_ToolShape) especially item #9 "Select the Sketch in the [Tree view](https://wiki.freecad.org/Tree_view). This ensures that the PropertyBag created in the next step will be nested in the Body" ?

I just tested with a custom shape I already have made and single clicked on PropBag, then right clicked & added property (deselected the "Prefix Group Name Options, entered details, created property, then in Tree added value. Saved and closed the Shape, created a new ToolBit based on that shape and the new property and value were present in the ToolBit.

I have a memory that @sliptonic did explain some detailed part of creating a shape recently, but I cannot find that, only the post that led to details I linked above. Because I knew there was some specific step in creating Tool Shape from scratch, I took shortcut and just edited an existing default shape to create the ones I required.

oodanner, 2024-09-12:
I added a video to file.io at this link so you can see exactly the process I'm using to generate the error and possibly be able to recreate it yourself.  https://file.io/l6Cs28au4XTa

oodanner, 2024-09-12:
Just to clarify, this has nothing to do with where the PropertyBag is nested as I know it gets nested under the body.  Perhaps some cleanup would be to make sure in the programming that the PropertyBag attributes always get nested properly in the body and also to just add a Diameter property with Length value to the PropertyBag by default since it won't work without at least that basic property.  I know this isn't a oft used tool by a lot of people but, I find being able to make custom tools to be very handy and I think with a little more work in this area more people would use it and possibly start contributing to some sort of user maintained tool database.  All I can say is that I started dabbling with FreeCAD a few years ago and it always left me wanting something else.  Can't say how much I appreciate the continued development on it and the solid progress you guys have all made.  I think I can finally ditch all of my other modeling tools now with RC1.

sliptonic, 2025-10-16:
I've tested with new tool shapes after the migration.  The property pane and the propertybag dialog seem to be operating correctly.  I'm closing this but will reopen if I've misunderstood the problem.