FreeCAD/FreeCAD #16442: Poor performance of viewport after upgrading to 1.0RC

skrystosik, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have Mac pro 5.1 with 2x xeons 128gb ram, xr6600xt grapihc card running osx monterey. On FreeCAD 0.22 dev build I had no performance issues. It was using GPU properly. I upgraded to 1.0 RC and nomatter if I pick in display settings using GPU or software openGL or any other settings. Viewport performance is poor 20-30 FPS. On 0.22 had 60 FPS + without any issue. 

### Full version info

```shell
OS: macOS 12.7.6
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.92.0
  * QuickMeasure 2022.10.28
  * fasteners 0.5.28
  * 3D_Printing_Tools
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-11:
Related : https://github.com/FreeCAD/FreeCAD/issues/16426 both reports appear to be Conda builds if that's relevant.

skrystosik, 2024-09-11:
Yes indeed, seems to be the same case. I was looking for but using different key.

luzpaz, 2024-09-11:
Let's close in favor of #16426

closed 2024-09-11 by luzpaz