FreeCAD/FreeCAD #16445: Activating a PartDesign's body switches to part design workbench

Media-Evil, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am using an assembly workbench with custom toolbars and I don't want FreeCAD to switch me over to PartDesign workbench every time I activate a body (make it an active body). It is incredibly irritating. This thing must be optional, not the default.

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.5.28
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Zolko-123, 2024-09-11:
@luzpaz added Type: Bug

this is no "Bug" it's how it's supposed to be. 

@Media-Evil :

> with custom toolbars

put your custom toolbars **also** into the _PartDesign_ workbench. That way, when you activate the Body and FreeCAD switches to the PartDesign WB as it's supposed-to, you'll have your custom tools available

mwganson, 2024-09-29:
There is a parameter for this action, but you must create it for yourself and set it to false.  Tools menu -> edit parameters -> BaseApp -> Preferences -> Mod -> PartDesign -> create a new boolean named "SwitchToWB" and set it to false.

luzpaz, 2024-09-30:
@mwganson Nice trick! Maybe we can make this a little more prominent somehow ? Any ideas?

mwganson, 2024-09-30:
It's in the wiki here:

https://wiki.freecad.org/Fine-tuning/en

in the part design section.

maxwxyz, 2026-11-07:
Part of this is in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/82d48774b4a99944f54dee2fb541992e40d94ef5

There is a checkbox in preferences to turn off the workbench switch. It's still enabled by default.