FreeCAD/FreeCAD #16446: [FC RC1]: Changing the main window in tile placement mode (windows change places).

kaiwas, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce the problem.
Open the file.
Place windows in tile mode.
Try switching windows. If everything is fine, switch the beam workbench to the techdro workbench.

The window positions change and switch between workbenches.

Sometimes the problem disappears if another workbench is selected.

Video demonstration.

https://github.com/user-attachments/assets/c5aad95b-9f24-4745-a82b-1b479b0e218c

The test file is attached (remove the zip extension)
[Doors_plan.FCStd.zip](https://github.com/user-attachments/files/16963213/Doors_plan.FCStd.zip)

This problem occurs when changing other modes (for example, spreadsheets)

https://github.com/user-attachments/assets/a2c97ed6-2601-405f-bf86-23f77e3db7fa

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.9.1
  * PitchedRoof
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-09-11:
Hi,

I can confirm the issue. I used empty new file, tiled two windows a switch between BIM and TechDraw workbench. After switching between workbenches and view windows they switch.

Plus I get this FreeCAD error output:
`14:53:08  Traceback (most recent call last):
14:53:08    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_38643\Mod\BIM\bimcommands\BimViews.py", line 285, in update
14:53:08      if hasattr(v.ViewObject, "Proxy"):
14:53:08                 ^
14:53:08  UnboundLocalError: cannot access local variable 'v' where it is not associated with a value
`

FreeCAD version:
`OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
`

luzpaz, 2024-09-11:
> 14:53:08  Traceback (most recent call last): 14:53:08    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_38643\Mod\BIM\bimcommands\BimViews.py", line 285, in update 14:53:08      if [...]

@Roy-043 any clue why BIM error occurs ?

kaiwas, 2024-09-11:
the problem is definitely not in the bim.
This problem occurs with any windows of different types. (at the end I gave an example with spreadsheets, but here is an example with techdraw and spreadsheets)
I think this will also appear in the draft workbench.

https://github.com/user-attachments/assets/c19211b8-0327-48a7-be5e-b5f34f5ae193

semhustej, 2024-09-11:
The other issue with the BIM view is independent of this one, I will create a new issue.

kaiwas, 2025-02-27:
This setting affects the problem. When it is active, when selecting tabs, switching between workbenches occurs. But at the same time, the windows change their position

[image]

luzpaz, 2025-05-11:
Is this relevant to 1.0 and 1.1dev ?

kaiwas, 2025-05-12:
> Is this relevant to 1.0 and 1.1dev ?

I checked it in Version: 1.1.0dev.41628 (Git) AppImage. 

Tile windows behavior has improved (they don't get replaced).
But there are some minor issues with switching workbenches when I select windows. But it's not critical.
It's much better.

https://github.com/user-attachments/assets/f9bbc779-ab62-4bbb-9677-cb3c24b41a8e

semhustej, 2025-09-21:
The issue is still valid.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.2

```

FEA-eng, 2026-08-25:
Still relevant (at least with the "Remember active workbench by tab" setting and when clicking on windows with BIM and Spreadsheet.

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260720 (Git shallow)
Build date: 2026/07/20 16:51:42
Build type: Release
Branch: grafted,grafted
Hash: 057d51f846f535f4f0b48492e586be0a76b9efe2
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```