FreeCAD/FreeCAD #16454: List Property Editor has hard coded colors, so it is impossible to manage by Themes

mnesarco, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Editor dialog for List based properties has colors set by c++ code. It makes it impossible to customize based on theme.

[image]

Code is here: https://github.com/FreeCAD/FreeCAD/blob/614a14fe87af6ea0a978ee6312b298e4033f1839/src/Gui/Widgets.cpp#L1359

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * DynamicData 2.62.0
  * MnesarcoUtils 0.2.12
  * sheetmetal 0.4.25
  * A2plus 0.4.67
  * fcscript 0.0.1
  * asm4 0.50.7
  * Curves 0.6.42
  * ProDarkThemePreferencePack 1.0.0
  * Marz 0.1.8
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-09-14:
Oh yeah. This feature is a real eyesore. White text on a yellow background is a separate torture.

mnesarco, 2024-09-18:
I still don't know how to make this themable, but at least I can make it text black over yellow background as a temporal workaround (it will work on light or dark modes). Would you accept that change as a PR?

[image]

[image]

luzpaz, 2024-09-18:
It's a good start!

MisterMakerNL, 2024-09-18:
Light or dark modes? Are you talking about the themes or your linux theme setting?

mnesarco, 2024-09-18:
> Light or dark modes? Are you talking about the themes or your linux theme setting?

It does not matter, as selection colors are fixed by code, black over yellow is readable no matter the freecad theme o linux theme. currently the code sets background as yellow no matter which color is text, this workaround sets text color to black so it is always readable over yellow. At least until this can be made customizable by themes.

mnesarco, 2024-09-18:
Wait a minute, apparently I have found a better solution that makes it theme-able. Or at least not hard coded colors.
[image]
[image]

MisterMakerNL, 2024-09-19:
That ladt solution would be nice, as stylesheets use accent color 1 as selection color. And general text color is also set to white or black. Having it hardcoded here while every other Qlabel aka textbox just uses these colors is a bit wrong.

closed 2024-09-20 by chennes