FreeCAD/FreeCAD #16456: editing model causes 3d model in 3d view to go black

brettowe, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when editing models on windows the 3d view of the model will go black
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Anything else?

for reference the view after closing the file and reopening (expected)
[image]
closing freecad itself is not required

the model seen as black/white is grey when rotated slightly
[image]

I've seen this several different files so perhaps only require operation is a pocket on a pad

OS is windows 10
graphics card is an amd radeon rx 6600xt

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-12:
@brettowe could you upload the zipped FCStd file? With which version was this file created? Does it work normally with the stable 0.21.2 version?

brettowe, 2024-09-12:
i'm not sharing any of the files that ive had it happen in, I will attempt to make an example file.
quick tests has shown its perhaps not as easy to trigger as i had first indicated
all files in common are multi body files and the shader gets messed up after a pocket operation most often.
and I don't recall ever seeing this happen in 0.21 series

closed 2025-07-23 by brettowe