FreeCAD/FreeCAD #16458: Inexisting Sub Folders

Fouleur, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Dans le menu  Édition -> Préférences n'existe pas

### Full version info

```shell
Dans l'aide il est dit : 
En définir un manuellement dans le menu Édition -> Préférences -> Import-Export -> DWG

Le menu préférences n'existe pas ??

Merci pour votre aide.
```

### Subproject(s) affected?

File formats

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-11:
Please translate to english, Thx

Roy-043, 2024-09-11:
https://wiki.freecad.org/Preferences_Editor/fr
> Pour lancer l'Éditeur de préférences, sélectionnez l'option Édition → Préférences... dans le menu. Sur macOS, l'option Préférences... apparaît dans le menu FreeCAD à la place (c'est l'emplacement par défaut des préférences sur Mac).

closed 2024-09-11 by Roy-043