FreeCAD/FreeCAD #16459: UI:UX: The new freeze object is gone.

leoheck, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some days ago I found this icon right-clicking on an object in the Objects tree.
Right now I found out how to use it, but I cannot locate this icon anymore. 
Is this a bug?
I would like to use this new feature.

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-09-11:
Hi, a known bug hide the `Toggle freeze` menu item under Part Design. It is shown when you switch to another workbench.
Listed in https://github.com/FreeCAD/FreeCAD/issues/12829

leoheck, 2024-09-11:
Thanks @FlachyJoe , is there any workaround? If yes, then I will close this issue.

closed 2024-09-11 by Roy-043