FreeCAD/FreeCAD #16467: Assembly: fit selection doesn't work

MiqCG, 2024-09-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When in an Assembly, View Fit selection (V,S) doesn't work, nor by keystrokes nor by mouse clicking.
On the other hand, Fit all (V, F) works fine. I have checked that Fit selection works correctly in this revision, on a Part design object

Maybe related to #14007.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * fasteners 0.5.27
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-12:
@PaddleStroke is this a dupe ?

PaddleStroke, 2024-09-12:
I cannot reproduce. Can you please provide a file and reproduction steps?

MiqCG, 2024-09-12:
Yes, of course. See attached assembly file (Ass_2) and video.

https://github.com/user-attachments/assets/d2c00e75-1e5d-465f-aa0c-abd6902bc6e5

[Measure.zip](https://github.com/user-attachments/files/16977917/Measure.zip)

PaddleStroke, 2024-09-12:
I have run some tests. This is not an assembly issue. This is an App::Link issue.
Just make a simple App::Link of Cos for example and try and you have the same issue.

On top of that please note that it is broken only if you have a face/edge/vertex selected. If you select the whole App::Link object in the tree, then it works correctly.

MiqCG, 2024-09-12:
Thanks for the investigation. 
I have updated the title and "Subproject(s) affected?".

PaddleStroke, 2025-07-22:
still relevant

PaddleStroke, 2025-09-22:
> On top of that please note that it is broken only if you have a face/edge/vertex selected. If you select the whole App::Link object in the tree, then it works correctly.

It appears that fit selection on elements (face edges vertex) is also broken for non-link. So it appears that it is not specific to Links.

FEA-eng, 2026-02-24:
I can't reproduce it, but there's a PR linked to this issue, so let's leave it open, I guess.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260217 (Git shallow)
Build date: 2026/02/17 21:26:14
Build type: Release
Branch: (HEAD detached at 3bad5dabd)
Hash: 3bad5dabde38f1a901fa1475e30f5caee7f9846a
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```